NabuCasa/universal-silabs-flasher #3: Flasher hangs after successful flash - doesn't reset device

tube0013, 2022-11-11:
Debog log minus most of the xmodem chunks.

```$ universal-silabs-flasher -vvv --device socket://192.168.1.33:6638 --bootloader-baudrate 115200 --baudrate 115200 flash --firmware MGM210PA32JIA_ncp-uart-sw_6.10.3_TubesZB_2022.gbl```

<details>
  <summary>Command output</summary>
```
2022-11-11 14:08:53 mercury asyncio[2332491] DEBUG Using selector: EpollSelector
2022-11-11 14:08:53 mercury universal_silabs_flasher.flasher[2332491] INFO Probing ApplicationType.GECKO_BOOTLOADER
2022-11-11 14:08:53 mercury universal_silabs_flasher.common[2332491] DEBUG Connection made: <_SelectorSocketTransport fd=3 read=idle write=<idle, bufsize=0>>
2022-11-11 14:08:53 mercury universal_silabs_flasher.common[2332491] DEBUG Sending data b'3'
2022-11-11 14:08:55 mercury universal_silabs_flasher.flasher[2332491] INFO Probing ApplicationType.CPC
2022-11-11 14:08:55 mercury universal_silabs_flasher.common[2332491] DEBUG Connection made: <_SelectorSocketTransport fd=7 read=idle write=<idle, bufsize=0>>
2022-11-11 14:08:55 mercury universal_silabs_flasher.cpc[2332491] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-11-11 14:08:55 mercury universal_silabs_flasher.common[2332491] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-11-11 14:08:56 mercury universal_silabs_flasher.cpc[2332491] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2022-11-11 14:08:56 mercury universal_silabs_flasher.cpc[2332491] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-11-11 14:08:56 mercury universal_silabs_flasher.common[2332491] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-11-11 14:08:57 mercury universal_silabs_flasher.cpc[2332491] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2022-11-11 14:08:57 mercury universal_silabs_flasher.cpc[2332491] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-11-11 14:08:57 mercury universal_silabs_flasher.common[2332491] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-11-11 14:08:58 mercury universal_silabs_flasher.cpc[2332491] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2022-11-11 14:08:58 mercury universal_silabs_flasher.cpc[2332491] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-11-11 14:08:58 mercury universal_silabs_flasher.common[2332491] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-11-11 14:08:59 mercury universal_silabs_flasher.cpc[2332491] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2022-11-11 14:08:59 mercury universal_silabs_flasher.flasher[2332491] INFO Probing ApplicationType.EZSP
2022-11-11 14:08:59 mercury asyncio[2332491] DEBUG Using selector: EpollSelector
2022-11-11 14:09:00 mercury bellows.ezsp[2332491] DEBUG Resetting EZSP
2022-11-11 14:09:00 mercury bellows.uart[2332491] DEBUG Resetting ASH
2022-11-11 14:09:00 mercury bellows.uart[2332491] DEBUG Sending: b'1ac038bc7e'
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2022-11-11 14:09:01 mercury bellows.ezsp.protocol[2332491] DEBUG Send command version: (4,)
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG Sending: b'004221a850ed2c7e'
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG Data frame: b'0142a1a85d2825c334c57e'
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG Sending: b'8160597e'
2022-11-11 14:09:01 mercury bellows.ezsp.protocol[2332491] DEBUG Application frame 0 (version) received: b'09023071'
2022-11-11 14:09:01 mercury bellows.ezsp[2332491] DEBUG Switching to EZSP protocol version 9
2022-11-11 14:09:01 mercury bellows.ezsp.protocol[2332491] DEBUG Send command version: (9,)
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG Sending: b'7d314221a9542a1c9cfe7e'
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG Data frame: b'1242a1a9542a1cb069e5931e7e'
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG Sending: b'82503a7e'
2022-11-11 14:09:01 mercury bellows.ezsp.protocol[2332491] DEBUG Application frame 0 (version) received: b'09023071'
2022-11-11 14:09:01 mercury bellows.ezsp[2332491] DEBUG EZSP Stack Type: 2, Stack Version: 7130, Protocol version: 9
2022-11-11 14:09:01 mercury bellows.ezsp.protocol[2332491] DEBUG Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-11-11 14:09:01 mercury bellows.uart[2332491] DEBUG Sending: b'224321a9fe2a1683467e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Data frame: b'2343a1a9fe2a15b32670b97e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Sending: b'83401b7e'
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Application frame 170 (getValue) received: b'00017f'
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Send command getConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>,)
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Sending: b'334021a9062a14a80d7e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Data frame: b'3440a1a9062a153259c0337e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Sending: b'8430fc7e'
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Application frame 82 (getConfigurationValue) received: b'008000'
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Free/configured buffers before any configurations: 127/128
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Sending: b'444121a9072a38b359c1ad7e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Data frame: b'4541a1a9072a15cd287e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Sending: b'8520dd7e'
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Application frame 83 (setConfigurationValue) received: b'00'
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Send command setConfigurationValue: (<EzspConfigId.CONFIG_MULTICAST_TABLE_SIZE: 6>, 16)
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Sending: b'554621a9072a7d33a259f81d7e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Data frame: b'5646a1a9072a208c667e'
2022-11-11 14:09:02 mercury bellows.uart[2332491] DEBUG Sending: b'8610be7e'
2022-11-11 14:09:02 mercury bellows.ezsp.protocol[2332491] DEBUG Application frame 83 (setConfigurationValue) received: b'35'
[... cut ...] \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x18\xe9' (attempt 0)
2022-11-11 14:11:23 mercury universal_silabs_flasher.xmodemcrc[2332491] DEBUG Got response: b'\x06'
2022-11-11 14:11:23 mercury universal_silabs_flasher.xmodemcrc[2332491] DEBUG Firmware upload progress: 100.00%
2022-11-11 14:11:23 mercury universal_silabs_flasher.xmodemcrc[2332491] DEBUG Sending data b'\x04' (attempt 0)
2022-11-11 14:11:23 mercury universal_silabs_flasher.xmodemcrc[2332491] DEBUG Got response: b'\x06'
```

MattWestb, 2022-11-11:
@tube0013 In the command output the ` ``` `  shall being on one blank line or its not working.

tube0013, 2022-11-11:
thanks, I'm pretty sure I had that but the edit seemed to have fixed it.

puddly, 2022-11-11:
Is there nothing else after the final `DEBUG Got response: b'\x06'`?

puddly, 2022-11-11:
Also, does this still happen over USB?

tube0013, 2022-11-11:
USB works fine.

<details>
  <summary>ending USB</summary>

```
2022-11-11 16:13:31 mercury universal_silabs_flasher.xmodemcrc[2342577] DEBUG Got response: b'\x06'
2022-11-11 16:13:31 mercury universal_silabs_flasher.xmodemcrc[2342577] DEBUG Firmware upload progress: 100.00%
2022-11-11 16:13:31 mercury universal_silabs_flasher.xmodemcrc[2342577] DEBUG Sending data b'\x04' (attempt 0)
2022-11-11 16:13:31 mercury universal_silabs_flasher.xmodemcrc[2342577] DEBUG Got response: b'\x06'
2022-11-11 16:13:31 mercury universal_silabs_flasher.common[2342577] DEBUG Received data b'\r\nSerial upload complete\r\n\x00\r\nGec'
2022-11-11 16:13:31 mercury universal_silabs_flasher.gecko_bootloader[2342577] DEBUG Parsing State.WAITING_UPLOAD_DONE: bytearray(b'\r\nSerial upload complete\r\n\x00\r\nGec')
2022-11-11 16:13:31 mercury universal_silabs_flasher.gecko_bootloader[2342577] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGec')
2022-11-11 16:13:31 mercury universal_silabs_flasher.common[2342577] DEBUG Received data b'ko Bootloader v2.01.01\r\n1. uploa'
2022-11-11 16:13:31 mercury universal_silabs_flasher.gecko_bootloader[2342577] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.01.01\r\n1. uploa')
2022-11-11 16:13:31 mercury universal_silabs_flasher.common[2342577] DEBUG Received data b'd gbl\r\n2. run\r\n3. ebl info\r\nBL >'
2022-11-11 16:13:31 mercury universal_silabs_flasher.gecko_bootloader[2342577] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.01.01\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL >')
2022-11-11 16:13:31 mercury universal_silabs_flasher.common[2342577] DEBUG Received data b' \x00'
2022-11-11 16:13:31 mercury universal_silabs_flasher.gecko_bootloader[2342577] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.01.01\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2022-11-11 16:13:31 mercury universal_silabs_flasher.gecko_bootloader[2342577] DEBUG Detected version string '2.01.01'
2022-11-11 16:13:31 mercury universal_silabs_flasher.common[2342577] DEBUG Sending data b'2'
```
</details>

TCP flash just hangs after ```DEBUG Got response: b'\x06'```

I just did a manual flash over tcp with serial app on my mac, and once it finished the efr32 didn't reset or go back the bootloader menu, I had to manually reset it to start the ncp

puddly, 2022-11-11:
Did you try sending `3`? Perhaps the bootloader is waiting for input but the prompt was never printed.

tube0013, 2022-11-11:
sending ```2```  after the xmodem transfer completes starts up the ncp

puddly, 2022-11-11:
It seems like the TCP server is losing data after the XMODEM transfer completes. Mine did the same thing but at least printed the second half of the menu.

tube0013, 2022-11-11:
I connected a second terminal (top) to it for the xmodem transfer and then was able to get the full output on the 1st terminal (bottom):

[image]

puddly, 2022-11-11:
The serial connection isn't closed during the bootloader session so the debug log contains everything that was sent.

Try the latest `dev`. If you are able to get the application to launch even though there is no prompt, that means the bootloader is still listening. This new version works around this problem on all of the hardware configurations I have available.

tube0013, 2022-11-11:
okay pulled dev 5 mins ago, and tried, seems to have read the menu, but still hangs and doesn't disconnect serial.

```
022-11-11 17:40:03 mercury universal_silabs_flasher.xmodemcrc[2349687] DEBUG Firmware upload progress: 100.00%
2022-11-11 17:40:03 mercury universal_silabs_flasher.xmodemcrc[2349687] DEBUG Sending data b'\x04' (attempt 0)
2022-11-11 17:40:03 mercury universal_silabs_flasher.xmodemcrc[2349687] DEBUG Got response: b'\x06'
2022-11-11 17:40:03 mercury universal_silabs_flasher.common[2349687] DEBUG Sending data b'3'
2022-11-11 17:40:03 mercury universal_silabs_flasher.common[2349687] DEBUG Received data b'\r\nGecko Bootloader v2.01.01\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00'
2022-11-11 17:40:03 mercury universal_silabs_flasher.gecko_bootloader[2349687] DEBUG Parsing State.WAITING_UPLOAD_DONE: bytearray(b'\r\nGecko Bootloader v2.01.01\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
```

puddly, 2022-11-11:
I think there's something funky going on with the TCP server. It stalls during upload like 20% of the time (fixed by a physical reset) and upon reconnect, I receive three queued up `\x18` bytes (XMODEM `CAN`) as part of the first packet.

I've been unable to replicate this issue at all when connecting over Ethernet as opposed to WiFi:

```Python
Received data b'\r\nSerial upload complete\r\n\x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00'
                    ^^^^^^^^^^^^^^^^^^^^^^
                 #  what your bootloader seems to not want to send
```

tube0013, 2022-11-11:
which is really weird as it was clearly sent when I did the separate connection for xmodem.  bu you can see the connection that the xmodem was using shows no completion or menu after the flash but the other one does.

puddly, 2022-11-12:
Fixed by f8fada4b215e407c687685429dbc4b6613065a26

closed 2022-11-12 by puddly