NabuCasa/universal-silabs-flasher #9: Not possible flashing with defect APP

MattWestb, 2022-12-18:
If the APP (NCP/RCP) on the chip then its not working or is not  flashed and the chip is starting in bootloader mode the flasher cant flash.
```
(311) PS C:\msys32\Portainer\belloows\311> universal-silabs-flasher -v  --device socket://192.168.2.60:8888 flash  --firmware ncp-uart-hw-gp-6103.gbl
2022-12-18 09:34:14 MW-E3222 asyncio[16708] DEBUG Using proactor: IocpProactor
2022-12-18 09:34:14 MW-E3222 universal_silabs_flasher.flasher[16708] INFO Probing ApplicationType.GECKO_BOOTLOADER
2022-12-18 09:34:14 MW-E3222 universal_silabs_flasher.common[16708] DEBUG Connection made: <_ProactorSocketTransport fd=1020>
2022-12-18 09:34:14 MW-E3222 universal_silabs_flasher.common[16708] DEBUG Sending data b'3'
2022-12-18 09:34:16 MW-E3222 universal_silabs_flasher.flasher[16708] INFO Probing ApplicationType.CPC

CPC probing is deleted then is long and if failing as expected.

2022-12-18 09:34:21 MW-E3222 universal_silabs_flasher.flasher[16708] INFO Probing ApplicationType.EZSP
2022-12-18 09:34:21 MW-E3222 asyncio[16708] DEBUG Using proactor: IocpProactor
2022-12-18 09:34:22 MW-E3222 bellows.ezsp[16708] DEBUG Resetting EZSP
2022-12-18 09:34:22 MW-E3222 bellows.uart[16708] DEBUG Resetting ASH
2022-12-18 09:34:22 MW-E3222 bellows.uart[16708] DEBUG Sending: b'1ac038bc7e'
2022-12-18 09:34:27 MW-E3222 bellows.uart[16708] DEBUG Connection lost: None
2022-12-18 09:34:27 MW-E3222 bellows.uart[16708] DEBUG Closed serial connection
Error: Failed to probe running application type
```
Here is one tuya ZBGW with only bootloader after erasing the chip and its booting the bootloader then the APP is not prescient.
USF is sending `3` for rebooting the APP and then trying rebooting the APP (EZSP) to boot loader.
Then the APP is not loading OK is the USF thinking it cant flashing the new firmware.
Instead is shall being better sending `1` then probing bootloader and if getting `C` every second start sending the file.

This way is possible recover devices that have working bootloader but the APP is not  OK and having bootloader pin implanted like most USB devices is having by forcing bootloader mode.

By the way RCP looks working OK but i have problem getting the upload working OK but its one other story (likely problem with  the serial server on the ZBGW).
```
2022-12-18 09:08:38 MW-E3222 universal_silabs_flasher.flasher[14352] INFO Detected ApplicationType.CPC, version 4.2.0
2022-12-18 09:08:38 MW-E3222 universal_silabs_flasher.flash[14352] INFO Detected running firmware ApplicationType.CPC, version 4.2.0
```

puddly, 2022-12-18:
Can you run with `-vv` and post the full output? I think it should re-enter the bootloader after trying to start.

MattWestb, 2022-12-18:
Flash erase and flashing only bootloader = not starting APP/ APP is backing out (not craching) or the user forcing bootloader mode with external button:
```
(311) PS C:\msys32\Portainer\belloows\311> universal-silabs-flasher -vv  --device socket://192.168.2.60:8888 flash  --firmware .\ncp-uart-hw-gp-6103B.gbl
2022-12-18 18:26:00 MW-E3222 asyncio[19204] DEBUG Using proactor: IocpProactor
2022-12-18 18:26:00 MW-E3222 universal_silabs_flasher.flasher[19204] INFO Probing ApplicationType.GECKO_BOOTLOADER
2022-12-18 18:26:00 MW-E3222 universal_silabs_flasher.common[19204] DEBUG Connection made: <_ProactorSocketTransport fd=996>
2022-12-18 18:26:00 MW-E3222 universal_silabs_flasher.common[19204] DEBUG Sending data b'3'
2022-12-18 18:26:02 MW-E3222 universal_silabs_flasher.flasher[19204] INFO Probing ApplicationType.CPC
2022-12-18 18:26:02 MW-E3222 universal_silabs_flasher.common[19204] DEBUG Connection made: <_ProactorSocketTransport fd=996>
2022-12-18 18:26:02 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-12-18 18:26:02 MW-E3222 universal_silabs_flasher.common[19204] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-12-18 18:26:03 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2022-12-18 18:26:04 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-12-18 18:26:04 MW-E3222 universal_silabs_flasher.common[19204] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-12-18 18:26:05 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2022-12-18 18:26:05 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-12-18 18:26:05 MW-E3222 universal_silabs_flasher.common[19204] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-12-18 18:26:06 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2022-12-18 18:26:06 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2022-12-18 18:26:06 MW-E3222 universal_silabs_flasher.common[19204] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2022-12-18 18:26:07 MW-E3222 universal_silabs_flasher.cpc[19204] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2022-12-18 18:26:07 MW-E3222 universal_silabs_flasher.flasher[19204] INFO Probing ApplicationType.EZSP
2022-12-18 18:26:07 MW-E3222 asyncio[19204] DEBUG Using proactor: IocpProactor
2022-12-18 18:26:08 MW-E3222 bellows.ezsp[19204] DEBUG Resetting EZSP
2022-12-18 18:26:08 MW-E3222 bellows.uart[19204] DEBUG Resetting ASH
2022-12-18 18:26:08 MW-E3222 bellows.uart[19204] DEBUG Sending: b'1ac038bc7e'
2022-12-18 18:26:13 MW-E3222 bellows.uart[19204] DEBUG Connection lost: None
2022-12-18 18:26:13 MW-E3222 bellows.uart[19204] DEBUG Closed serial connection
Error: Failed to probe running application type
(311) PS C:\msys32\Portainer\belloows\311>
```
Firs sending `3` if the chip is in bootloader mode => booting to APP (EZSP) and continuing probing RCP and EZSP.

If start sending `1` (start x-modem transfer) and getting `C` every second we is in bootloader mode and can start sending the file.

MattWestb, 2022-12-18:
Edit: The the bootloader is getting one `3` is chip is trying starting the APP but is not flashed so its returning to the bootloader. 

We are missing 2 scenarios: Forced bootloader of the user and no loading APP but working bootloader.

puddly, 2022-12-18:
> Firs sending `3` if the chip is in bootloader mode => booting to APP (EZSP) and continuing probing RCP and EZSP.

`3` should be "ebl info":

```
Gecko Bootloader v1.2.3
1. upload gbl
2. run
3. ebl info
BL > 
```

My understanding is that you can run `3` as many times as you want, since it just causes the bootloader prompt to be printed again each time. If `3` is sent and no response from the bootloader is received, the chip is not in the bootloader, no?

MattWestb, 2022-12-18:
I think  you is being right !!
I have trying 3 before but if having the full screen with bootloader menu you dont see its being updated.

I was thinking 3 was run (2) = running  the APP.

I dont knowing if 3 / EBL inf is working OK in all bootloader versions i think some have not function if treading Silabs papers (it was for old EBL files not GBL).

So is My bootloader not sending one OK boot menu or is the USF not detecting it ?

MattWestb, 2022-12-18:
Im very sorry but i was having problem with  sending to the bootloader.
Then using one IKEA MG21 module and flashing only bootloader its looks like this:
```
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.flash[16484] INFO Detected running firmware ApplicationType.GECKO_BOOTLOADER, version 2.00.02
MG210L-610-ncp-uart-hw-gp-multi-railHW.gbl
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.common[16484] DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x000002677C3009D0>, Serial<id=0x2677c2e55a0, open=True>(port='com8', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.common[16484] DEBUG Sending data b'3'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.common[16484] DEBUG Received data b'\r\nGecko Bootloader v2.00.02\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.gecko_bootloader[16484] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.00.02\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.gecko_bootloader[16484] DEBUG Detected version string '2.00.02'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.common[16484] DEBUG Sending data b'3'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.common[16484] DEBUG Received data b'\r\nGecko Bootloader v2.00.02\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.gecko_bootloader[16484] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.00.02\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.gecko_bootloader[16484] DEBUG Detected version string '2.00.02'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.common[16484] DEBUG Sending data b'1'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.common[16484] DEBUG Received data b'\r\nbegin upload\r\n\x00C'
2022-12-18 20:09:12 MW-E3222 universal_silabs_flasher.gecko_bootloader[16484] DEBUG Parsing State.WAITING_XMODEM_READY: bytearray(b'\r\nbegin upload\r\n\x00C')
2022-12-18 20:09:13 MW-E3222 universal_silabs_flasher.xmodemcrc[16484] DEBUG Sending data b'\x01\x01\xfe\xeb\x17\xa6\x03\x08\x00\x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\xf4\n\n\xf4\x1c\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfd\x03\x03\xfd\xb0\x00\x00\x00\x00@\x00\x00\xa0c\x00 m\xf1\x00\x00\xf1\xaa\x00\x00\xf5\xaa\x00\x00\xa7\n\n\x01\x00B\x00\x00\x04\x18\x0f\xac0j)\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x90\xca\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\xbd\x17' (attempt 0)
```
So USF is catching chips in booloader mode and can flashing them without problem (if having the comport working OK !!!!).

So also possible for user recovering there device if getting them in bootloader mode :-)))

Its real **"PUDDLY"** !!!!

Thanks for support !!

closed 2022-12-18 by MattWestb

MattWestb, 2022-12-18:
Bay the way how to putting comports with number over 9 in windows ?
com11 is not working but remapping it to com8 its working OK.
msys32 need special "tagging" for comports over 9 to working but i cant finding how to do it OK.

puddly, 2022-12-18:
That is interesting, I didn't know there was a different syntax for 10+...

It looks like it is `\\.\COM10` (https://support.microsoft.com/en-us/topic/howto-specify-serial-ports-larger-than-com9-db9078a5-b7b6-bf00-240f-f749ebfd913e). I will make a PR to fix this.

MattWestb, 2022-12-18:
Great !!!
its the same syntax as for msys32 i can saying then i see it :-))

Windows 10 was having some very strange bugs then deleting used comports in GUI and the system was away not releasing the comport number so i was having com47 before hacking the register for getting the blocking away (and was deleting little too much).

It shall being fixed in current Window 10.

MattWestb, 2022-12-18:
```
bellows -d \\.\com11 -b 115200  info
[2c:11:65:ff:fe:b6:d8:c8]
[0xfffe]
[<EmberNetworkStatus.NO_NETWORK: 0>]
[<EmberStatus.NOT_JOINED: 147>, <EmberNodeType.undefined_0x71: 113>, EmberNetworkParameters(extendedPanId=00:00:00:00:00:00:00:00, panId=0xffff, radioTxPower=255, radioChannel=0, joinMethod=<EmberJoinMethod.USE_MAC_ASSOCIATION: 0>, nwkManagerId=0xffff, nwkUpdateId=0, channels=<Channels.NO_CHANNELS: 0>)]
[<EmberStatus.NOT_JOINED: 147>, EmberCurrentSecurityState(bitmask=<EmberCurrentSecurityBitmask.STANDARD_SECURITY_MODE: 0>, trustCenterLongAddress=00:d0:00:00:49:5f:80:00)]
Manufacturer: IKEA of Sweden
Board name: Marcus NCP by MW
EmberZNet version: 6.10.3.0 build 297
```
Need reconnect the USB-Serial after changing  the mapping of the comport in advanced setting of the comport in device manager for getting it working OK :-)))