NabuCasa/universal-silabs-flasher #17: Can't flash anything to a SkyConnect 1.0

Djelibeybi, 2023-02-09:
I think I've bricked my SkyConnect because all attempts to flash a firmware just fail with the following output:

```shell
universal-silabs-flasher --verbose --device COM7 flash --firmware silabs-firmware\EmberZNet\NabuCasa_SkyConnect_EZSP_v7.1.1.0_ncp-uart-hw_115200.gbl --allow-cross-flashing --allow-downgrades --allow-reflash-same-version
2023-02-09 20:38:50 DjeliRGB asyncio[4200] DEBUG Using proactor: IocpProactor
2023-02-09 20:38:50 DjeliRGB universal_silabs_flasher.flasher[4200] INFO Probing ApplicationType.GECKO_BOOTLOADER
2023-02-09 20:38:50 DjeliRGB universal_silabs_flasher.common[4200] DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x0000019F6610F390>, Serial<id=0x19f660e9840, open=True>(port='COM7', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2023-02-09 20:38:50 DjeliRGB universal_silabs_flasher.common[4200] DEBUG Sending data b'3'
2023-02-09 20:38:52 DjeliRGB universal_silabs_flasher.flasher[4200] INFO Probing ApplicationType.CPC
2023-02-09 20:38:52 DjeliRGB universal_silabs_flasher.common[4200] DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0x0000019F66118410>, Serial<id=0x19f660e9840, open=True>(port='COM7', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2023-02-09 20:38:52 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-02-09 20:38:52 DjeliRGB universal_silabs_flasher.common[4200] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-02-09 20:38:53 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2023-02-09 20:38:53 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-02-09 20:38:53 DjeliRGB universal_silabs_flasher.common[4200] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-02-09 20:38:54 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2023-02-09 20:38:54 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-02-09 20:38:54 DjeliRGB universal_silabs_flasher.common[4200] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-02-09 20:38:55 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2023-02-09 20:38:55 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2023-02-09 20:38:55 DjeliRGB universal_silabs_flasher.common[4200] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2023-02-09 20:38:56 DjeliRGB universal_silabs_flasher.cpc[4200] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2023-02-09 20:38:56 DjeliRGB universal_silabs_flasher.flasher[4200] INFO Probing ApplicationType.EZSP
2023-02-09 20:38:56 DjeliRGB bellows.ezsp[4200] DEBUG Resetting EZSP
2023-02-09 20:38:56 DjeliRGB bellows.uart[4200] DEBUG Resetting ASH
2023-02-09 20:38:56 DjeliRGB bellows.uart[4200] DEBUG Sending: b'1ac038bc7e'
2023-02-09 20:39:01 DjeliRGB bellows.uart[4200] DEBUG Connection lost: None
2023-02-09 20:39:01 DjeliRGB bellows.uart[4200] DEBUG Closed serial connection
Error: Failed to probe running application type
```

Is there any way to recover?

puddly, 2023-02-09:
Were you previously able to flash firmware? If so, which one did you flash?

Djelibeybi, 2023-02-09:
I tried to flash the Nabu Casa `RCPMultiPAN` firmware a couple of weeks ago. It _seemed_ to work, but the process either crashed or aborted and I haven't been able to get the SkyConnect to work since.

puddly, 2023-02-09:
Do you remember which exact firmware it was?

Can you also try doing this via the web flasher? Scroll to the bottom of https://skyconnect.home-assistant.io/

Djelibeybi, 2023-02-09:
I don't recall which exact firmware it was. To be honest, once it appeared to be bricked, I ignored the issue for a while until I had more time to try and poke at it. I'll give the web flasher a whirl in a couple of hours.

Djelibeybi, 2023-02-09:
The web installer failed as well. Here's the `silabs_flasher.log` from the attempt, though it's content is identical to the previous output.

[silabs_flasher.log](https://github.com/NabuCasa/universal-silabs-flasher/files/10700901/silabs_flasher.log)

puddly, 2023-02-09:
Is this using the same operating system/platform as before? Do you have a second computer to try this on?

If you are using macOS, make sure to use `/dev/cu.SLAB_USBtoUART` and not the other serial port.

Djelibeybi, 2023-02-09:
I've tried it on Windows 11, in Ubuntu using WSL2 and usbipd but not macOS yet. I shall give that a whirl now.

Djelibeybi, 2023-02-09:
Looks like it had the same error on macOS (Ventura, 13.2 on an M1 Pro)

[silabs_flasher.log](https://github.com/NabuCasa/universal-silabs-flasher/files/10701275/silabs_flasher.log)

puddly, 2023-02-09:
Just to confirm, you see two serial ports when connecting to the SkyConnect, correct? Due to a driver bug, the one labeled SkyConnect doesn't actually work:

[image]

If not, install the driver (https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads) and connect to the `CP210x USB to UART ...` port.

puddly, 2023-02-09:
If you can figure out *exactly* which firmware you flashed to it, I may be able to help.  Otherwise, it unfortunately seems like your SkyConnect is bricked...

Djelibeybi, 2023-02-09:
Do you know if there's a physical way to switch to the bootloader? If it's already bricked, it wouldn't do any more damage to open it up and poke at a contact.

Djelibeybi, 2023-02-09:
I got all excited because I found some flash commands in the history on the Raspberry Pi but it turned out to be the flashing I did to my Sonoff Zigbee 3.0 Dongle-E, not the SkyConnect.

Djelibeybi, 2023-02-16:
Fixed. The solution was to change the baud rate to 230400, as found by anshulgupta0803 in this Home Assistant forum thread: https://community.home-assistant.io/t/problems-reflashing-my-skyconnect-usb-dongle-back-to-emberznet/510244/10?u=djelibeybi

closed 2023-02-16 by Djelibeybi

puddly, 2023-02-16:
That's really unusual, we don't ship a single firmware with a 230400 baudrate. Do you happen to know exactly what firmware you flashed to get it into this state?