NabuCasa/universal-silabs-flasher #18: [Future request] Light device info without flashing firmware.

MattWestb, 2023-02-09:
Then cooking different firmware types in different device like tuya / LIDL ZBGW and ESPHome serial server device and local connected one its no easy way fast checking if the NCP / RCP is working OK on the chip / module. 
Its possible doing with bellows / zigpy CLI but it being much easier and better if that was possible with the flasher.

If only getting the EZSP protocol version its running /  bootloader string / the build of the RCP (depends of the mode the chip is booted in) you can see the APP on the chip is alive and can moving on and doing more tests in ZHA and so on.

If it only taking the first working alternative its OK but better restarting from the app (EZSP / RCP) and getting the bootloader string wold being great.

Thanks in advance for great coding and making our system better and better !!!

darkxst, 2023-06-24:
I've added a `probe` command that will show the running app and version string. For EZSP it also dumps Ember config. It does not however reboot and probe bootloader.

MattWestb, 2023-06-24:
Thanks @darkxst i have seen it but not testing it.
I think its great function then testing different firmware and like have one fast check if the SOC is up and running or is not starting the app.

Im closing this issue then it shall being implanted and working.

Thanks one more tome for the work you have doing !!

closed 2023-06-24 by MattWestb