NabuCasa/universal-silabs-flasher #19: Unable to update SkyConnect firmware on https://skyconnect.home-assistant.io/

zachberger, 2023-02-10:
Hi There,

After reading the recent blog post, I thought I'd try updating the firmware on my skyconnect. When I attempt on the website I see the following message: "The running firmware could not be detected."

[image]

I've attached the debug logs.

[silabs_flasher.log](https://github.com/NabuCasa/universal-silabs-flasher/files/10710628/silabs_flasher.log)

puddly, 2023-02-10:
Are you running macOS? If so, I think you are affected by https://github.com/NabuCasa/sl-web-tools/pull/4.

I'll copy/paste the text here:

> The built-in drivers on macOS do not work properly with the Home Assistant SkyConnect. Install the updated [Silicon Labs CP2102 driver](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads) and re-connect to the serial port titled **CP210x USB to UART Bridge Controller (cu.SLAB_USBtoUART)**

zachberger, 2023-02-10:
Thats it, thank you

closed 2023-02-10 by zachberger