NabuCasa/universal-silabs-flasher #33: Error trying to flash SkyConnect to Zigbee only firmware

MiloRoNu, 2023-04-06:
I enabled the multiprotocol support on the SkyConnect, but it is behaving erratically, so I’m trying to revert the firmware using the universal flasher, but it ends up with an Atribute Error.

I’m trying to flash it on a Raspberry Pi with Debian 64-bit and running Home Assistant Supervised.

This is the error that shows up:

Traceback (most recent call last):
File "/home/pi/.local/bin/universal-silabs-flasher", line 8, in 
sys.exit(main())
File "/usr/lib/python3/dist-packages/click/core.py", line 829, in call
return self.main(*args, **kwargs)
File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
rv = self.invoke(ctx)
File "/usr/lib/python3/dist-packages/click/core.py", line 1256, in invoke
Command.invoke(self, ctx)
File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/usr/lib/python3/dist-packages/click/core.py", line 610, in invoke
return callback(*args, **kwargs)
File "/usr/lib/python3/dist-packages/click/decorators.py", line 21, in new_func
return f(get_current_context(), *args, **kwargs)
File "/home/pi/.local/lib/python3.9/site-packages/universal_silabs_flasher/flash.py", line 151, in main
if ctx.get_parameter_source("baudrate") != click.core.ParameterSource.DEFAULT:
AttributeError: 'Context' object has no attribute 'get_parameter_source'

puddly, 2023-04-06:
It looks like the version of your `click` package is too old.

Try installing everything in an isolated virtual environment:

```console
$ virtualenv -p 3 venv
$ source venv/bin/activate
$ pip install universal-silabs-flasher
$ # flash
$ deactivate
```

Alternatively, use the web flasher: https://skyconnect.home-assistant.io/firmware-update/

MiloRoNu, 2023-04-06:
Thank you, I’ll try it when I get home. I don’t know why, but the web flasher doesn’t work with my unit. And I tried with to computers, with different os and browsers.

MiloRoNu, 2023-04-06:
It worked perfectly! Thank you very much!

closed 2023-04-06 by MiloRoNu