NabuCasa/universal-silabs-flasher #35: ZB-GW04 fails to enter bootloader

darkxst, 2023-04-19:
I just got a ZB-GW04 v1.2 for testing.
This is currently installed with factory EZSP 6.7.9 firmware.

Flashing fails when trying to enter Gecko Bootloader. Device **does** enter bootloader (Which can confirm with serial console), but apparently doesnt reply with expected response. Flasher then crashing.

Here are the logs
```
2023-04-19 11:35:21 xps15 universal_silabs_flasher.flash[203873] INFO Cross-flashing from FirmwareImageType.NCP_UART_HW to FirmwareImageType.RCP_UART_802154
2023-04-19 11:35:21 xps15 bellows.ezsp[203873] DEBUG Resetting EZSP
2023-04-19 11:35:21 xps15 bellows.uart[203873] DEBUG Resetting ASH
2023-04-19 11:35:21 xps15 bellows.uart[203873] DEBUG Sending: b'1ac038bc7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command version: (4,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'004221a850ed2c7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'0142a1a85c2885d52df87e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'8160597e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received version: [8, 2, 26512]
2023-04-19 11:35:22 xps15 bellows.ezsp[203873] DEBUG Switching to EZSP protocol version 8
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command version: (8,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'7d314221a9542a1d8cdf7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'1242a1a9542a1db0c9f38a237e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'82503a7e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received version: [8, 2, 26512]
2023-04-19 11:35:22 xps15 bellows.ezsp[203873] DEBUG EZSP Stack Type: 2, Stack Version: 6790, Protocol version: 8
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'224321a9fe2a1683467e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'2343a1a9fe2a15b361489a7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'83401b7e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received getValue: [<EzspStatus.SUCCESS: 0>, b'8']
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command getConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'334021a9062a14a80d7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'3440a1a9062a15f259d6677e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'8430fc7e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received getConfigurationValue: [<EzspStatus.SUCCESS: 0>, 64]
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Free/configured buffers before any configurations: 56/64
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'444121a9fe2a167d3a397e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'4541a1a9fe2a15b36165917e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'8520dd7e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received getValue: [<EzspStatus.SUCCESS: 0>, b'8']
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command getConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'554621a9062a1437d37e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'5646a1a9062a15f25968ee7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'8610be7e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received getConfigurationValue: [<EzspStatus.SUCCESS: 0>, 64]
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Free/configured buffers before all memory allocation: 56/64
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'664721a9fe2a1699cd7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'6747a1a9fe2a15b36184c17e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'87009f7e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received getValue: [<EzspStatus.SUCCESS: 0>, b'8']
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command getConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'774421a9062a14b2867e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Data frame: b'7044a1a9062a15f2591a3c7e'
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'8070787e'
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Application frame received getConfigurationValue: [<EzspStatus.SUCCESS: 0>, 64]
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Free/configured buffers after all memory allocation: 56/64
2023-04-19 11:35:22 xps15 bellows.ezsp.protocol[203873] DEBUG Send command launchStandaloneBootloader: (1,)
2023-04-19 11:35:22 xps15 bellows.uart[203873] DEBUG Sending: b'004521a9db2a144dc67e'
2023-04-19 11:35:27 xps15 bellows.uart[203873] DEBUG Connection lost: None
2023-04-19 11:35:27 xps15 bellows.uart[203873] DEBUG Closed serial connection
Traceback (most recent call last):
  File "/home/tim/anaconda3/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
  File "/home/tim/anaconda3/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/tim/anaconda3/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/tim/anaconda3/lib/python3.8/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/tim/anaconda3/lib/python3.8/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/tim/anaconda3/lib/python3.8/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/tim/anaconda3/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/tim/anaconda3/lib/python3.8/site-packages/universal_silabs_flasher/flash.py", line 34, in inner
    return asyncio.run(f(*args, **kwargs))
  File "/home/tim/anaconda3/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/home/tim/anaconda3/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/tim/anaconda3/lib/python3.8/site-packages/universal_silabs_flasher/flash.py", line 318, in flash
    await flasher.enter_bootloader()
  File "/home/tim/anaconda3/lib/python3.8/site-packages/universal_silabs_flasher/flasher.py", line 278, in enter_bootloader
    res = await ezsp.launchStandaloneBootloader(0x01)
  File "/home/tim/anaconda3/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

```

puddly, 2023-04-19:
Indeed, it looks like the device fails to respond with the expected response:

[image]

This can probably be fixed by assuming the bootloader has launched after no response is received:

```diff
diff --git a/universal_silabs_flasher/flasher.py b/universal_silabs_flasher/flasher.py
index 2733e18..c9adc12 100644
--- a/universal_silabs_flasher/flasher.py
+++ b/universal_silabs_flasher/flasher.py
@@ -285,12 +285,18 @@ class Flasher:
                     await spinel.enter_bootloader()
         elif self.app_type is ApplicationType.EZSP:
             async with self._connect_ezsp(self.app_baudrate) as ezsp:
-                res = await ezsp.launchStandaloneBootloader(0x01)
-
-                if res[0] != bellows.types.EmberStatus.SUCCESS:
-                    raise RuntimeError(
-                        f"EmberZNet could not enter the bootloader: {res[0]!r}"
+                try:
+                    res = await ezsp.launchStandaloneBootloader(0x01)
+                except asyncio.TimeoutError:
+                    _LOGGER.warning(
+                        "Application failed to respond to bootloader launching command."
+                        " Assuming bootloader has launched."
                     )
+                else:
+                    if res[0] != bellows.types.EmberStatus.SUCCESS:
+                        raise RuntimeError(
+                            f"EmberZNet could not enter the bootloader: {res[0]!r}"
+                        )
         else:
             raise RuntimeError(f"Invalid application type: {self.app_type}")
```

darkxst, 2023-04-19:
I will try that.

Just for reference this is the response (tested with the elelabs util which also fails)
```
2023/04/19 12:19:59 Elelabs_EzspFwUtility:   launchStandaloneBootloader: STANDALONE_BOOTLOADER_NORMAL_MODE
2023/04/19 12:19:59 Elelabs_EzspFwUtility:   [ EZSP  REQUEST ] 02 00 01 8f 00 01
2023/04/19 12:20:02 Elelabs_EzspFwUtility:   [ EZSP RESPONSE ] 02 80 01 8f d0 18 b8 1e f1 29 4e c5 75 d0 26 f3 3a 4b c4 8c aa 02 f9 dd b0 52 a7 c5 50 0d aa e1 fc f0 4f fa 8f ab b4 b4 b6 49 eb 24 4f a4 e6 44 15 85 98 00 ec 4c 92 7f 08 33 d4 82 1c 18 75 60 61 d4 e8 c0 27 c6
```

darkxst, 2023-04-19:
That patch works and flash completed successfully.

PR #36 (2023-04-20): Continue after EZSP timeout after bootloader launch
Fixes #35
closed 2023-04-20 by puddly