NabuCasa/universal-silabs-flasher #39: Flasher won't talk to bootloader

clydebarrow, 2023-05-23:
I have a brand new HA Yellow, but the radio won't work. When I try to update the firmware using this command line:

```
universal-silabs-flasher -v --probe-method bootloader --device /dev/ttyAMA1 flash --allow-cross-flashing --firmware EmberZNet/NabuCasa_Yellow_EZSP_v6.10.3.0_PA32_ncp-uart-hw_115200.gbl
```

It clearly is exchanging data via the serial port, but never progresses:

```
2023-05-23 11:46:56 a0d7b954-ssh universal_silabs_flasher.common[1622] DEBUG Sending data b'3'
2023-05-23 11:46:57 a0d7b954-ssh universal_silabs_flasher.common[1622] DEBUG Received data b'\r\nGecko Bootload'
2023-05-23 11:46:57 a0d7b954-ssh universal_silabs_flasher.gecko_bootloader[1622] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootload')
2023-05-23 11:46:57 a0d7b954-ssh universal_silabs_flasher.common[1622] DEBUG Received data b'er v2.00.01\r\n1. '
2023-05-23 11:46:57 a0d7b954-ssh universal_silabs_flasher.gecko_bootloader[1622] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.00.01\r\n1. ')
2023-05-23 11:46:57 a0d7b954-ssh universal_silabs_flasher.common[1622] DEBUG Received data b' v2.00.01\r\n1. up'
2023-05-23 11:46:57 a0d7b954-ssh universal_silabs_flasher.gecko_bootloader[1622] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.00.01\r\n1.  v2.00.01\r\n1. up')
```

darkxst, 2023-05-28:
@clydebarrow Try add   `--yellow-gpio-reset` to the end of your command.

clydebarrow, 2023-05-28:
> Try add --yellow-gpio-reset

I already did - no difference.

puddly, 2023-05-30:
Are you positive no addons or integrations are running that could also be talking to the serial port? ZHA? Z2M? Silicon Labs Multiprotocol addon?

clydebarrow, 2023-05-30:
The only integration installed that would talk to it AFAIK is the `Yellow Multi-PAN
Zigbee Home Automation` which was pre-installed when I got the unit.  There are no other integrations showing that mention Zigbee or SiLabs and I certainly have not installed anything else related to Zigbee.

puddly, 2023-05-30:
Multi-PAN is not pre-installed with the Yellow, you would have had to enable it manually within the Hardware menu. This is likely the problem you're facing, since the addon is simultaneously trying to talk to the same device.

You can follow the instructions in the Yellow documentation to disable multi-PAN: https://yellow.home-assistant.io/guides/disable-multiprotocol/

clydebarrow, 2023-05-30:
It is possible I enabled the Multi-PAN while trying to get the radio to work.

I will note that since I opened this issue I had determined that the radio was working (with Multi-PAN) it just seems to have limited range. I moved my target Zigbee device right next to the HAYellow and it then found it.

I just now followed the steps to disable multi-PAN and it seems to be working ok. Thanks for the help. I will note that I had looked at the Yellow documentation pages and didn't find anything useful, maybe some extra entries in the FAQ pointing to the Multiprotocol guides would help.

closed 2023-05-30 by clydebarrow

puddly, 2023-05-30:
> I moved my target Zigbee device right next to the HAYellow and it then found it.

Do you see a startup warning about channel utilization? If so, your Yellow may be in a poor position and too close to a potential interference source. The firmware doesn't really impact range in any way.

clydebarrow, 2023-05-31:
No, I have made sure that my WiFi channels are not overlapping the Zigbee channel 11 (and there are no other WiFi sources nearby.)

I was trying to update the firmware just to eliminate that as a cause of the non-connection problem, but I think now I just had unrealistic expectations of the Zigbee range. I think I will just stick with WiFi since I have good coverage from multiple APs.