NabuCasa/universal-silabs-flasher #44: Unable to install the universal-silabs-flasher

Lumute, 2023-08-05:
Hi,

I want to install the Zigbee Only firmware on my Home Assistant Yellow, it is running latest OS version 10.4.

When I run the installation command I get this error:

```
➜  ~ pip install universal-silabs-flasher
Collecting universal-silabs-flasher
  Downloading universal_silabs_flasher-0.0.13-py3-none-any.whl (43 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.3/43.3 kB 1.6 MB/s eta 0:00:00
Collecting click (from universal-silabs-flasher)
  Downloading click-8.1.6-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.9/97.9 kB 3.2 MB/s eta 0:00:00
Collecting zigpy (from universal-silabs-flasher)
  Downloading zigpy-0.56.4-py3-none-any.whl (178 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 178.3/178.3 kB 4.1 MB/s eta 0:00:00
Collecting crc (from universal-silabs-flasher)
  Downloading crc-4.3.0-py3-none-any.whl (7.7 kB)
Collecting bellows>=0.34.3 (from universal-silabs-flasher)
  Downloading bellows-0.35.8-py3-none-any.whl (234 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 234.2/234.2 kB 5.2 MB/s eta 0:00:00
Collecting coloredlogs (from universal-silabs-flasher)
  Downloading coloredlogs-15.0.1-py2.py3-none-any.whl (46 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.0/46.0 kB 2.7 MB/s eta 0:00:00
Collecting async-timeout (from universal-silabs-flasher)
  Downloading async_timeout-4.0.2-py3-none-any.whl (5.8 kB)
Collecting typing-extensions (from universal-silabs-flasher)
  Downloading typing_extensions-4.7.1-py3-none-any.whl (33 kB)
Collecting gpiod (from universal-silabs-flasher)
  Downloading gpiod-1.5.4-py3-none-any.whl (30 kB)
Collecting click-log>=0.2.1 (from bellows>=0.34.3->universal-silabs-flasher)
  Downloading click_log-0.4.0-py2.py3-none-any.whl (4.3 kB)
Collecting pure-pcapy3==1.0.1 (from bellows>=0.34.3->universal-silabs-flasher)
  Downloading pure-pcapy3-1.0.1.tar.gz (5.9 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting voluptuous (from bellows>=0.34.3->universal-silabs-flasher)
  Downloading voluptuous-0.13.1-py3-none-any.whl (29 kB)
Requirement already satisfied: setuptools in /usr/lib/python3.11/site-packages (from pure-pcapy3==1.0.1->bellows>=0.34.3->universal-silabs-flasher) (67.7.2)
Collecting attrs (from zigpy->universal-silabs-flasher)
  Downloading attrs-23.1.0-py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.2/61.2 kB 3.5 MB/s eta 0:00:00
Collecting aiohttp (from zigpy->universal-silabs-flasher)
  Downloading aiohttp-3.8.5-cp311-cp311-musllinux_1_1_aarch64.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 9.4 MB/s eta 0:00:00
Collecting aiosqlite>=0.16.0 (from zigpy->universal-silabs-flasher)
  Downloading aiosqlite-0.19.0-py3-none-any.whl (15 kB)
Collecting crccheck (from zigpy->universal-silabs-flasher)
  Downloading crccheck-1.3.0-py3-none-any.whl (21 kB)
Collecting cryptography (from zigpy->universal-silabs-flasher)
  Downloading cryptography-41.0.3-cp37-abi3-musllinux_1_1_aarch64.whl (4.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.2/4.2 MB 10.2 MB/s eta 0:00:00
Collecting pyserial-asyncio (from zigpy->universal-silabs-flasher)
  Downloading pyserial_asyncio-0.6-py3-none-any.whl (7.6 kB)
Collecting humanfriendly>=9.1 (from coloredlogs->universal-silabs-flasher)
  Downloading humanfriendly-10.0-py2.py3-none-any.whl (86 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 86.8/86.8 kB 3.2 MB/s eta 0:00:00
Collecting charset-normalizer<4.0,>=2.0 (from aiohttp->zigpy->universal-silabs-flasher)
  Downloading charset_normalizer-3.2.0-cp311-cp311-musllinux_1_1_aarch64.whl (191 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 191.7/191.7 kB 5.8 MB/s eta 0:00:00
Collecting multidict<7.0,>=4.5 (from aiohttp->zigpy->universal-silabs-flasher)
  Downloading multidict-6.0.4-cp311-cp311-musllinux_1_1_aarch64.whl (141 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 141.0/141.0 kB 4.5 MB/s eta 0:00:00
Collecting yarl<2.0,>=1.0 (from aiohttp->zigpy->universal-silabs-flasher)
  Downloading yarl-1.9.2-cp311-cp311-musllinux_1_1_aarch64.whl (248 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 248.8/248.8 kB 5.3 MB/s eta 0:00:00
Collecting frozenlist>=1.1.1 (from aiohttp->zigpy->universal-silabs-flasher)
  Downloading frozenlist-1.4.0-cp311-cp311-musllinux_1_1_aarch64.whl (252 kB)
[... 91 lines cut ...]
      copying cffi/setuptools_ext.py -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/lock.py -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/api.py -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/vengine_gen.py -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/ffiplatform.py -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/__init__.py -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/_cffi_include.h -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/parse_c_type.h -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/_embedding.h -> build/lib.linux-aarch64-cpython-311/cffi
      copying cffi/_cffi_errors.h -> build/lib.linux-aarch64-cpython-311/cffi
      running build_ext
      building '_cffi_backend' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/c
      gcc -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DFFI_BUILDING=1 -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.11 -c c/_cffi_backend.c -o build/temp.linux-aarch64-cpython-311/c/_cffi_backend.o
      error: command 'gcc' failed: No such file or directory
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cffi
Successfully built pure-pcapy3
Failed to build cffi
ERROR: Could not build wheels for cffi, which is required to install pyproject.toml-based projects

```

darkxst, 2023-08-05:
> I want to install the Zigbee Only firmware on my Home Assistant Yellow, it is running latest OS version 10.4.

If you are running this directly from HA OS, the easiest way is to install the [Silabs Flasher Addon](https://github.com/home-assistant/addons/blob/master/silabs_flasher/DOCS.md). It will update your Yellow to the latest Zigbee firmware!

Lumute, 2023-08-05:
> f you are running this directly from HA OS, the easiest way is to install the [Silabs Flasher Addon](https://github.com/home-assistant/addons/blob/master/silabs_flasher/DOCS.md). It will update your Yellow to the latest Zigbee firmware!

Awesome, thanks! that did the trick... although it says the latest version is 2.0 with firmware v7.3.1.0, yet it isnatlled 0.1.1 and that flashed 7.2.2.0, but its a Zigbee onlu firmware anyway, I guess 2.0 will be available later...

This should be included in the official instructions, this is liked from the Yellow hardware option to switch the firmware:

https://github.com/NabuCasa/silabs-firmware/wiki/Flash-Silicon-Labs-radio-firmware-manually

darkxst, 2023-08-05:
> although it says the latest version is 2.0 with firmware v7.3.1.0

There was a glitch with upload, it will be available soon though!
https://github.com/home-assistant/addons/pull/3156#issuecomment-1666342088

> Should I do anything to clean my yellow from that half installation?

I dont think so, `pip` should have cleaned up after the failed build.

Lumute, 2023-08-05:
Great, thank you!

darkxst, 2023-08-06:
> This should be included in the official instructions, this is linked from the Yellow hardware option to switch the firmware:
> 
> https://github.com/NabuCasa/silabs-firmware/wiki/Flash-Silicon-Labs-radio-firmware-manually

To install directly on HA OS probably needs this command: 
```
pip install \
   --find-links "https://wheels.home-assistant.io/musllinux/" \
     universal-silabs-flasher
 ```    
Perhaps that wiki page should be updated as such.

puddly, 2023-09-22:
This package is now included with HA Core. It's also available via the Silicon Labs Flasher addon, which supports custom URLs.

closed 2023-09-22 by puddly