NabuCasa/universal-silabs-flasher #46: Exception is thrown when trying to flash

danielBreitlauch, 2023-11-24:
Command used:
```
$ universal-silabs-flasher --device /dev/ttyUSB0 flash --firmware NabuCasa_SkyConnect_OpenThread_RCP_v2.3.2.0_ot-rcp_hw_460800.gbl
```

Exception:
```
Traceback (most recent call last):
  File "/usr/local/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1134, in invoke
    Command.invoke(self, ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/universal_silabs_flasher/flash.py", line 153, in main
    if ctx.get_parameter_source("baudrate") != click.core.ParameterSource.DEFAULT:
AttributeError: 'Context' object has no attribute 'get_parameter_source'
```

The machine is an Ubuntu 20.4 on x86.

puddly, 2023-11-24:
This sounds like a dependency problem.  How did you install the package? Try it in a fresh virtualenv.

danielBreitlauch, 2023-11-24:
Now I got further along. I did it this way now:

```
$ pipenv shell
$ pip install universal-silabs-flasher
$ pipenv run universal-silabs-flasher --device /dev/ttyUSB0 flash --firmware NabuCasa_SkyConnect_OpenThread_RCP_v2.3.2.0_ot-rcp_hw_460800.gbl
```

And got a new exception:
```
2023-11-24 17:17:54 barkeeper universal_silabs_flasher.flash[3056305] INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.3.2', ezsp_version=None, ot_rcp_version='SL-OPENTHREAD/2.3.2.0_GitHub-e6df00dd6' (2.3.2.0), fw_type=<FirmwareImageType.OT_RCP: 'ot-rcp'>, baudrate=460800)
2023-11-24 17:17:54 barkeeper universal_silabs_flasher.flasher[3056305] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-11-24 17:17:56 barkeeper universal_silabs_flasher.flasher[3056305] INFO Probing ApplicationType.SPINEL at 460800 baud
2023-11-24 17:18:00 barkeeper universal_silabs_flasher.flasher[3056305] INFO Probing ApplicationType.CPC at 460800 baud
2023-11-24 17:18:05 barkeeper universal_silabs_flasher.flasher[3056305] INFO Probing ApplicationType.CPC at 115200 baud
2023-11-24 17:18:09 barkeeper universal_silabs_flasher.flasher[3056305] INFO Probing ApplicationType.CPC at 230400 baud
2023-11-24 17:18:13 barkeeper universal_silabs_flasher.flasher[3056305] INFO Probing ApplicationType.EZSP at 115200 baud
Traceback (most recent call last):
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/universal_silabs_flasher/flash.py", line 36, in inner
    return asyncio.run(f(*args, **kwargs))
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/universal_silabs_flasher/flash.py", line 322, in flash
    await flasher.probe_app_type(
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/universal_silabs_flasher/flasher.py", line 166, in probe_ezsp
    async with self._connect_ezsp(baudrate) as ezsp:
  File "/usr/lib/python3.8/contextlib.py", line 171, in __aenter__
    return await self.gen.__anext__()
  File "/root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages/universal_silabs_flasher/emberznet.py", line 18, in connect_ezsp
    config.CONF_DEVICE_BAUDRATE: baudrate,
AttributeError: module 'bellows.config' has no attribute 'CONF_DEVICE_BAUDRATE'
```

puddly, 2023-11-24:
Ah, that is a bug. Thanks!

danielBreitlauch, 2023-11-24:
Installed dependencies: 
Collecting bellows>=0.34.3

Requirement already satisfied: setuptools in /root/.local/share/virtualenvs/testflash-6R93O_W4/lib/python3.8/site-packages (from pure-pcapy3==1.0.1->bellows>=0.34.3->universal-silabs-flasher) (44.0.0)

If that helps.

Baegi, 2023-11-24:
@danielBreitlauch I finally found the problem for this. The bellow release 0.37.0 removed the required CONF_DEVICE_BAUDRATE field. Run in your venv

`pip install bellows==0.36.8`

to force installation of the working version. Flashing worked flawlessly after that.

alsFC, 2023-11-24:
Got an error related to the same reason today. This was the stack trace:

`Traceback (most recent call last):
  File "/usr/local/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/universal_silabs_flasher/flash.py", line 36, in inner
    return asyncio.run(f(*args, **kwargs))
  File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.10/site-packages/universal_silabs_flasher/flash.py", line 322, in flash
    await flasher.probe_app_type(
  File "/usr/local/lib/python3.10/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
  File "/usr/local/lib/python3.10/site-packages/universal_silabs_flasher/flasher.py", line 166, in probe_ezsp
    async with self._connect_ezsp(baudrate) as ezsp:
  File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/contextlib.py", line 199, in __aenter__
    return await anext(self.gen)
  File "/usr/local/lib/python3.10/site-packages/universal_silabs_flasher/emberznet.py", line 18, in connect_ezsp
    config.CONF_DEVICE_BAUDRATE: baudrate,
AttributeError: module 'bellows.config' has no attribute 'CONF_DEVICE_BAUDRATE'. Did you mean: 'CONF_DEVICE_PATH'?`

Setting bellows to 0.36.8 also worked for me.

danielBreitlauch, 2023-11-24:
@Baegi THX. That solved it and everything works.

darkxst, 2023-11-25:
@puddly this breaks the web flashers as well it seems.

puddly, 2023-11-25:
This has been fixed by https://github.com/zigpy/bellows/pull/597 so the existing web flashers now work again without needing modifications.

https://github.com/NabuCasa/universal-silabs-flasher/pull/47 is the proper fix for this problem, however, so update to `sl-web-tools==0.10.1` once it's released.

closed 2023-11-25 by puddly

darkxst, 2023-11-25:
Thanks puddly, will update my web flasher once its released!