NabuCasa/universal-silabs-flasher #48: SL Web Tools and universal-silabs-flasher  Unable to Flash SkyConnect

moutasem1989, 2023-11-25:
Trying through Terminal and through the Web Tool i keep getting an error to update the firmware on the SkyConnect.

In the terminal I receive this error when running the command:

```
$ universal-silabs-flasher --device /dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_d014f323acdbed11b03cf22d62c613ac-if00-port0     flash --allow-cross-flashing --firmware ~/silabs-firmware/RCPMultiPAN/NabuCasa_SkyConnect_RCP_v4.1.4_rcp-uart-hw-802154_115200.gbl
2023-11-25 18:24:10 homeserver universal_silabs_flasher.flash[90031] INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.1.4', ezsp_version=None, ot_rcp_version=None, fw_type=<FirmwareImageType.RCP_UART_802154: 'rcp-uart-802154'>, baudrate=None)
2023-11-25 18:24:10 homeserver universal_silabs_flasher.flasher[90031] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-11-25 18:24:12 homeserver universal_silabs_flasher.flasher[90031] INFO Probing ApplicationType.CPC at 460800 baud
2023-11-25 18:24:17 homeserver universal_silabs_flasher.flasher[90031] INFO Probing ApplicationType.CPC at 115200 baud
2023-11-25 18:24:21 homeserver universal_silabs_flasher.flasher[90031] INFO Probing ApplicationType.CPC at 230400 baud
2023-11-25 18:24:25 homeserver universal_silabs_flasher.flasher[90031] INFO Probing ApplicationType.EZSP at 115200 baud
2023-11-25 18:24:30 homeserver universal_silabs_flasher.flasher[90031] INFO Probing ApplicationType.SPINEL at 460800 baud
Error: Failed to probe running application type
```

Using the Web-Tool it throws this error:

```
Connection failed
The running firmware could not be detected.

Make sure the USB port works and if you are using a USB extension cable, make sure the cable can transfer data. Unplug the Home Assistant SkyConnect and plug it back in to reset and try again.
```

and the Log file is as following:

```
2023-11-25 18:36:32 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-11-25 18:36:32 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (115200 baudrate)
2023-11-25 18:36:32 emscripten universal_silabs_flasher.common[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1057e38>
2023-11-25 18:36:32 emscripten universal_silabs_flasher.common[42] DEBUG Sending data b'3'
2023-11-25 18:36:34 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2023-11-25 18:36:34 emscripten zigpy.serial[42] DEBUG Opening a serial connection to '/dev/webserial' (460800 baudrate)
```

My goal is to Enable multiprotocol support but I am ruunning HA in Docker. Anyone can help ?

puddly, 2023-11-25:
Have you disabled any possible integration that could be accessing the serial port? ZHA, Z2M, etc.

moutasem1989, 2023-11-25:
Thank you !
I did stop HA container but it worked only after stoping all running Docker Containers.
I did manage to update SkyConnect Zigbee