NabuCasa/universal-silabs-flasher #55: AttributeError: can_write_custom_eui64 not found in COMMANDS

IgnacioHR, 2023-12-29:
Yesterday I posted an issue [here](https://github.com/darkxst/silabs-firmware-builder/issues/38#issue-2058678911) but it might be the case there is a bug on this code.

After successfuly flashed `ncp-uart-hw-v7.3.2.0-zbdonglee-none-115200.gbl` I tried to execute `universal-silabs-flasher --device /dev/tty.usbserial-202305080921111 write-ieee --ieee e0798dfffe77c2f0` in order to write to the new dongle the same address I've in the old one and I received this error:

```
universal-silabs-flasher --device /dev/tty.usbserial-202305080921111 write-ieee --ieee e0798dfffe77c2f0
2023-12-28 18:28:43 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2023-12-28 18:28:45 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.CPC at 460800 baud
2023-12-28 18:28:49 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.CPC at 115200 baud
2023-12-28 18:28:54 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.CPC at 230400 baud
2023-12-28 18:28:58 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Probing ApplicationType.EZSP at 115200 baud
2023-12-28 18:28:59 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Detected ApplicationType.EZSP, version '7.3.2.0 build 212' (7.3.2.0.212) at 115200 baudrate (bootloader baudrate None)
2023-12-28 18:29:00 mbp-de-ignacio.local universal_silabs_flasher.flasher[20665] INFO Current device IEEE: e0:79:8d:ff:fe:d9:0e:a8
Traceback (most recent call last):
  File "/Users/ihr/zigbee-sonoff-firmware/venv/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/universal_silabs_flasher/flash.py", line 36, in inner
    return asyncio.run(f(*args, **kwargs))
  File "/Users/ihr/.asdf/installs/python/3.9.13/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Users/ihr/.asdf/installs/python/3.9.13/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/universal_silabs_flasher/flash.py", line 250, in write_ieee
    await ctx.obj["flasher"].write_emberznet_eui64(new_eui64)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/universal_silabs_flasher/flasher.py", line 353, in write_emberznet_eui64
    if not await ezsp.can_write_custom_eui64():
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/bellows/ezsp/__init__.py", line 300, in __getattr__
    return getattr(self._protocol, name)
  File "/Users/ihr/zigbee-sonoff-firmware/venv/lib/python3.9/site-packages/bellows/ezsp/protocol.py", line 132, in __getattr__
    raise AttributeError(f"{name} not found in COMMANDS")
AttributeError: can_write_custom_eui64 not found in COMMANDS
```

I'm not sure if the operation shall be supported or if the error shall be captured and turned into a more human readable error in any case I'm more interested in the process to virtually "duplicate" my coordinator on identical hardware in order to test with different firmware versions.

Regards
Ignacio

IgnacioHR, 2023-12-29:
Here it is
```
(venv) ihr@mbp-de-ignacio zigbee-sonoff-firmware % pip freeze
aiohttp==3.9.1
aiosignal==1.3.1
aiosqlite==0.19.0
async-timeout==4.0.3
attrs==23.1.0
bellows==0.37.4
cffi==1.16.0
click==8.1.7
click-log==0.4.0
coloredlogs==15.0.1
crc==6.0.0
crccheck==1.3.0
cryptography==41.0.7
frozenlist==1.4.1
humanfriendly==10.0
idna==3.6
multidict==6.0.4
pure-pcapy3==1.0.1
pycparser==2.21
pyserial==3.5
pyserial-asyncio==0.6
typing_extensions==4.9.0
universal-silabs-flasher==0.0.15
voluptuous==0.14.1
yarl==1.9.4
zigpy==0.60.2
(venv) ihr@mbp-de-ignacio zigbee-sonoff-firmware %
```

puddly, 2023-12-29:
Actually, never mind. You don't need to use the flasher for this.

Use zigpy-cli instead. The firmware you're running supports overwriting the IEEE address:

```bash
pip install zigpy-cli
zigpy -vvv radio --baudrate 115200 ezsp /dev/serial/by-id/OLD_RADIO backup -z > backup.json
zigpy -vvv radio --baudrate 115200 ezsp /dev/serial/by-id/NEW_RADIO restore backup.json

# Reset the old radio so that it doesn't interfere
zigpy -vvv radio --baudrate 115200 ezsp /dev/serial/by-id/OLD_RADIO reset

# Finally, compare backups
zigpy -vvv radio --baudrate 115200 ezsp /dev/serial/by-id/NEW_RADIO backup -z > backup2.json
diff backup.json backup2.json
```

IgnacioHR, 2023-12-29:
Thank you for the detailed information. I'm going to close this issue now.

closed 2023-12-29 by IgnacioHR