NabuCasa/universal-silabs-flasher #60: No firmware found when updating firmware

robarnold, 2024-01-04:
As a follow up to #57, I am getting further but still seeing issues when I try to update the bootloader. I was able to update the main application just fine. [Here](https://gist.github.com/robarnold/4a5a978c494f07f428d3a3fafbaa8861) are my logs from the attempt.

I assume there are two potential issues here:
1) "2.01.01" != "2.1.1" -> there should not be a firmware update attempt at all
2) some kind of user error (wrong file?), but hopefully the error message could be clearer if so

puddly, 2024-01-04:
1. The GBL image you are flashing contains no metadata so the flasher can't tell what firmware it contains, it will just flash it as requested.
2. The specific error is `Failed to boot`.

What specifically are you trying to do? Your bootloader is already `v2.01.01`, there's no need to flash `2.1.1` a second time. I suspect that the current bootloader will not allow you to flash a new bootloader if the new bootloader isn't an upgrade.

robarnold, 2024-01-04:
My goal is to set up a script to automatically flash the firmware if necessary on container start, before zigbee2mqtt starts running. This way I can just bump the file path to the new version (ideally using a tool like renovate) and redeploy the container the same way as an image bump. And hopefully I could revert firmware this way too if I run into issues. This orchestration process works for the application firmware for me right now, but not the bootloader firmware.

puddly, 2024-01-04:
I really would not mess with the bootloader like this. If the process is interrupted, you will be left with a completely bricked stick. There is almost never a reason to touch the bootloader: its only job is to launch the main application and write firmware updates. There haven't been any changes to the Gecko/Ember bootloader in many years, nor will there likely ever be for many more to come.

The error you're receiving is because the flasher expects to be launching an application after flashing a file. It looks like what happens after you flash the bootloader is that the main application is erased as well, so there is no application run. Hence the error.

robarnold, 2024-01-04:
Ok, it makes sense to me not to do it, thank you! It is a bit weird to me that the firmware update isn't idempotent and erases the application firmware. Should this be documented better?

closed 2024-01-04 by robarnold