NabuCasa/universal-silabs-flasher #66: [REQUEST] Support for Silicon Labs Z-Wave 700/800 OTW flashing of Z-Wave Controller

Hedda, 2024-04-08:
Requesting support in this "Universal Silicon Labs Flasher" tool for flashing Silabs Z-Wave 700 series and 800 series firmware via OTW,

I believe this is a relevant request here now that Nabu Casa has joined the Z-Wave Alliance and is working towards Z-Wave certification:

https://www.home-assistant.io/blog/2024/02/15/nabu-casa-joins-z-wave-alliance/

https://community.home-assistant.io/t/psa-upcoming-z-wave-js-changes-and-z-wave-certification-how-it-impacts-you/638288

https://www.home-assistant.io/integrations/zwave_js/

https://www.home-assistant.io/docs/z-wave/controllers/

So this is an enhancement feature request for Z-Wave OTW, to automatically enter bootloader mode and then flash firmware GBL image.

FYI, Silicon Labs already provides pre-built firmware image downloads for the Z-Wave 700/800 series chips via their Gecko SDK (GSDK):

https://github.com/SiliconLabs/gecko_sdk/

Example:

https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.4/protocol/z-wave/Apps/bin/gbl/

Please consider adding support to flashing OTW firmware ZW Serial API Controller GBL updates on all Z-Wave Controller USB sticks.

https://community.silabs.com/s/article/z-wave-700-otw-of-controller

https://community.silabs.com/s/article/z-wave-700-programming-uzb7-controller-stick

https://www.silabs.com/community/wireless/z-wave/knowledge-base.entry.html/2019/01/04/z-wave_700_programm-4Hx0

https://www.silabs.com/community/wireless/z-wave/forum.topic.html/will_serialapi_imagesupporttheserialbootloader-Wwib

The same OTW upgrade process and flash method for ZW SerialAPI Controller Firmware over UART serial interface should also be applicable to all adapters based on ZGM130/ZGM130S or ZGM230/ZGM230S radio modules and EFR32ZG14 or FR32ZG23 SoCs.

Note that Z-Wave 500 series OTW programming is slightly different from Z-Wave 700 and Z-Wave 800 series which is the same:

https://community.silabs.com/s/article/gateway-z-wave-500-how-to-implement-otw-programming

For reference, the Z-Wave JS UI app already has support for performing OTW (Over-The-Wire) firmware update of Z-Wave Controller.

https://zwave-js.github.io/node-zwave-js/#/troubleshooting/otw-upgrade?id=over-the-wire-otw-firmware-upgrades-of-z-wave-controllers

https://github.com/zwave-js/node-zwave-js/blob/master/docs/api/controller.md#updating-the-firmware-of-the-controller-otw

PS: Think this would enable forever goal of Home Assistant founders / Nabu Casa of "Streamlining Experiences" for HA frontend/GUI?

https://www.home-assistant.io/blog/2022/01/19/streamlining-experiences/

puddly, 2024-04-08:
Have you tested it?

Hedda, 2024-04-08:
> Have you tested it?

No I actually just assumed that it was not supported since there is no mention of support for the zwave ncp serial api. Is it now?

https://docs.silabs.com/z-wave/7.21.1/zwave-api/md-mnt-raid-workspaces-ws-y-vud-vssd9-overlay-gsdk-protocol-z-wave-apps-zwave-ncp-serial-api-readme