NabuCasa/universal-silabs-flasher #67: Dongle-E "Error: Failed to probe running application type"

majorsl, 2024-04-11:
I accidently flashed a 230400 baud rate file when it should have been a 115200 to my ZBDongle-E

The fix I found in another issue was to re-flash with the correct baud rate: https://github.com/NabuCasa/silabs-firmware/issues/13

Except, I can't.  It says the that baudrate has been deprecated.

If I try to flash the correct file, I get "Error: Failed to probe running application type" and honestly don't know how to fix this as there isn't much info.

puddly, 2024-04-11:
`--baudrate` has been deprecated because there isn't a *single* baudrate. Each application type can potentially use a different baudrate.

You don't need to specify a baudrate, however, because you don't need to trigger the bootloader from the application itself: your stick has a hardware bootloader trigger.

```console
universal-silabs-flasher \
    --device /dev/serial/by-id/... \
    --bootloader-reset sonoff \
    flash --force --firmware /path/to/fw.gbl
```

majorsl, 2024-04-12:
> `--baudrate` has been deprecated because there isn't a _single_ baudrate. Each application type can potentially use a different baudrate.
> 
> You don't need to specify a baudrate, however, because you don't need to trigger the bootloader from the application itself: your stick has a hardware bootloader trigger.
> 
> ```
> universal-silabs-flasher \
>     --device /dev/serial/by-id/... \
>     --bootloader-reset sonoff \
>     flash --force --firmware /path/to/fw.gbl
> ```

Ah, thank you!  I'm all set.

closed 2024-04-12 by majorsl