NabuCasa/universal-silabs-flasher #71: AttributeError: 'Context' object has no attribute 'get_parameter_source'

sinipelto, 2024-07-07:
With version 0.0.20

Tried following python versions:
3.8.10
3.9.5
3.12.4

Getting the following stactrace using 'probe' command:

python3.12 -m universal_silabs_flasher probe
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/pelto/.local/lib/python3.12/site-packages/universal_silabs_flasher/__main__.py", line 4, in <module>
    main()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1134, in invoke
    Command.invoke(self, ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pelto/.local/lib/python3.12/site-packages/universal_silabs_flasher/flash.py", line 170, in main
    if ctx.get_parameter_source("baudrate") != click.core.ParameterSource.DEFAULT:
       ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Context' object has no attribute 'get_parameter_source'

puddly, 2024-07-07:
Are you using the latest version of the `click` package?

sinipelto, 2024-07-08:
Problem solved. For me pip was offering click 7.0 as the latest version from pyPI. 

I manually upgraded click with:

python3.12 -m pip install click==8.1.7

Which resolved the issue.

Should still figure out whats the root cause for getting such old version automatically by installing the universal-silabs-flasher.

puddly, 2024-07-08:
I believe `get_parameter_source` was added only in 8.x so we should change pin the dependency's minimum version: `click>=8.0.0`

closed 2024-07-08 by puddly