NabuCasa/universal-silabs-flasher #73: Operation not permitted: '/dev/gpiochip0'

Charlespoole, 2024-07-10:
I'm unable to flash my on-board zigbee module on Homeassistant Yellow board. The error showed "Operation not permitted: '/dev/gpiochip0', **Here is the log:**
`2024-07-10 17:30:01.659 core-silabs-flasher universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.3.1', ezsp_version='7.3.1.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-07-10 17:30:01.660 core-silabs-flasher universal_silabs_flasher.flasher INFO Triggering yellow bootloader
Traceback (most recent call last):
  File "/usr/local/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())

  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)

  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)

  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))

  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)

  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)

  File "/usr/local/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flash.py", line 40, in inner
    return asyncio.run(f(*args, **kwargs))
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 654, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flash.py", line 341, in flash
    await flasher.probe_app_type()
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 173, in probe_app_type
    await self.enter_bootloader_reset(self._reset_target)
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 78, in enter_bootloader_reset
    await send_gpio_pattern(
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/gpio.py", line 125, in send_gpio_pattern
    await asyncio.get_running_loop().run_in_executor(
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/gpio.py", line 29, in _send_gpio_pattern
    chip = gpiod.chip(chip, gpiod.chip.OPEN_BY_PATH)
  File "/usr/local/lib/python3.11/site-packages/gpiod/libgpiodcxx/__init__.py", line 105, in __init__
    self.open(device, how)
  File "/usr/local/lib/python3.11/site-packages/gpiod/libgpiodcxx/__init__.py", line 132, in open
    chip_struct = func(device)
  File "/usr/local/lib/python3.11/site-packages/gpiod/libgpiod/__init__.py", line 105, in gpiod_chip_open
    fd = os_open(path, O_RDWR | O_CLOEXEC)
PermissionError: [Errno 1] Operation not permitted: '/dev/gpiochip0'`
Any clue why this is happening? End of the rope here, if you guys cannot help, might have to return my board for factory repair.

puddly, 2024-07-10:
I don't think this is a hardware issue.

What other addons do you have? Integrations?

Charlespoole, 2024-07-10:
> I don't think this is a hardware issue.
> 
> What other addons do you have? Integrations?

[image]
Here is my addons, no integration that could possibly use zigbee devices is installed.

puddly, 2024-07-10:
Just to confirm, you're using the [Silicon Labs Flasher Add-on](https://github.com/home-assistant/addons/tree/master/silabs_flasher)? I didn't notice it in your list of addons.

puddly, 2024-07-10:
That is likely the problem. Install it from the normal Home Assistant addon repository.

puddly, 2024-07-10:
> I also used the offical silabs-flasher addon, no luck there. 

Can you post a traceback of what happens when you try installing from the normal addon? Make sure to enable verbose mode:

[image]

puddly, 2024-07-10:
`Probing ApplicationType.GECKO_BOOTLOADER at 460800 baud` doesn't look right. Did you change the baudrate? I would switch it back to 115200.

Charlespoole, 2024-07-10:
> `Probing ApplicationType.GECKO_BOOTLOADER at 460800 baud` doesn't look right. Did you change the baudrate? I would switch it back to 115200.
Tried every baudrate listed, no luck.

puddly, 2024-07-10:
Let's get everything back to the defaults by uninstalling the addon and the container from Portainer.

Do you have ZHA or Z2M running? If so, they would prevent firmware flashing from succeeding. Disable them first. Afterwards, install the addon, enable verbose logging, pick `/dev/ttyAMA1`, and run it with the default config.

Charlespoole, 2024-07-11:
> Let's get everything back to the defaults by uninstalling the addon and the container from Portainer.
>
> Do you have ZHA or Z2M running? If so, they would prevent firmware flashing from succeeding. [...]

That did not work for me. I might have to factory reset the device this weekend. I will update the situation in this issue.

Charlespoole, 2024-07-13:
> That did not work for me. I might have to factory reset the device this weekend. I will update the situation in this issue.

Turns out factory resetting the device is not enough. I have to reinstall the entire OS to solve this problem.

closed 2024-11-05 by puddly