NabuCasa/universal-silabs-flasher #76: ValueError('Unknown firmware image type')

xabix, 2024-07-24:
Hello,

I am on a docker machine and installed the script with the pip command but I get this error message:

`universal-silabs-flasher 
Traceback (most recent call last):
  File "/home/docker/.local/bin/universal-silabs-flasher", line 5, in <module>
    from universal_silabs_flasher.__main__ import main
  File "/home/docker/.local/lib/python3.10/site-packages/universal_silabs_flasher/__main__.py", line 1, in <module>
    from .flash import main
  File "/home/docker/.local/lib/python3.10/site-packages/universal_silabs_flasher/flash.py", line 14, in <module>
    import bellows.types
  File "/home/docker/.local/lib/python3.10/site-packages/bellows/types/__init__.py", line 1, in <module>
    from .basic import *  # noqa: F401,F403
  File "/home/docker/.local/lib/python3.10/site-packages/bellows/types/basic.py", line 3, in <module>
    from zigpy.types import (  # noqa: F401
  File "/home/docker/.local/lib/python3.10/site-packages/zigpy/types/__init__.py", line 4, in <module>
    from .named import *  # noqa: F401,F403
  File "/home/docker/.local/lib/python3.10/site-packages/zigpy/types/named.py", line 8, in <module>
    import attrs
ModuleNotFoundError: No module named 'attrs'`

which I solved with:
`pip install --upgrade attrs`

when I do:
`universal-silabs-flasher --deviceuniversal-silabs-flasher --device /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221029103944-if00 flash --firmware ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl`

I get this issue:
`Error: 'ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl' does not appear to be a valid firmware image: ValueError('Unknown firmware image type')`

I tried with other images from: [https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee](url)

But same results. Any idea?

Merci

puddly, 2024-07-24:
Make sure the firmware files are actually firmware files and not HTML pages. The flasher has a complete file parser for every firmware type so it's not wrong when it says the firmware is invalid.

xabix, 2024-07-24:
wget needed the raw link to work properly
https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

closed 2024-07-24 by xabix