NabuCasa/universal-silabs-flasher #78: Unable to flash firmware to Sonoff E dongle - Failed to probe running application type

aquova, 2024-08-14:
All, 

I've been trying to update the firmware on my Sonoff dongle, but have been unable to do so, either from the web interface or the command line. When I attempt to do so, the program cycles through the various application types, fails to find a satisfactory one, then spits out the "Failed to probe running application type" error. This is on my desktop Linux machine, and my user is in the correct group for the USB device. Since this is my desktop machine, there is nothing running that would be using the dongle. 

Here is some of the error logs:

```
2024-08-14 12:59:00.914 sapphire asyncio DEBUG Using selector: EpollSelector
2024-08-14 12:59:00.914 sapphire universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-08-14 12:59:00.914 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0' (baudrate=115200, xonxoff=False, rtscts=False)
2024-08-14 12:59:00.937 sapphire universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x7bbe7f53c890>, Serial<id=0x7bbe7f5fa0e0, open=True>(port='/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-08-14 12:59:00.937 sapphire universal_silabs_flasher.common DEBUG Sending data b'\n'
2024-08-14 12:59:00.937 sapphire universal_silabs_flasher.common DEBUG Sending data b'3'
2024-08-14 12:59:02.951 sapphire universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2024-08-14 12:59:02.951 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0' (baudrate=460800, xonxoff=False, rtscts=False)
2024-08-14 12:59:02.975 sapphire universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0x7bbe7f53c740>, Serial<id=0x7bbe7f53feb0, open=True>(port='/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0', baudrate=460800, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-08-14 12:59:02.976 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-14 12:59:02.976 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-14 12:59:03.977 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2024-08-14 12:59:04.077 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-14 12:59:04.078 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-14 12:59:05.079 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2024-08-14 12:59:05.179 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-14 12:59:05.179 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-14 12:59:06.181 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2024-08-14 12:59:06.281 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-14 12:59:06.281 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-14 12:59:07.282 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2024-08-14 12:59:07.295 sapphire universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2024-08-14 12:59:07.295 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0' (baudrate=115200, xonxoff=False, rtscts=False)
2024-08-14 12:59:07.320 sapphire universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0x7bbe7f53caa0>, Serial<id=0x7bbe7f5fa0e0, open=True>(port='/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-08-14 12:59:07.320 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-14 12:59:07.320 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-14 12:59:08.321 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2024-08-14 12:59:08.421 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-14 12:59:08.422 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-14 12:59:09.422 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt [... cut ...]
2024-08-14 12:59:25.347 sapphire universal_silabs_flasher.spinel DEBUG Failed to send SpinelFrame(header=SpinelHeader(network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02'), trying again in 0.10s (attempt 4 of 4)
Error: Failed to probe running application type
```

Which was run via 
> $ universal-silabs-flasher -vv --device /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0 probe

I did also try to upgrade via my Linux laptop and an old MacBook, but didn't have any luck there either.

puddly, 2024-08-14:
According to the log, your coordinator doesn't respond to *any* known protocol at any baudrate so it's possible the firmware is corrupted and the application crashes. You can try to enter the bootloader with hardware instead of software:

```bash
universal-silabs-flasher -vv \
	--device /dev/serial/by-id/... \
	--bootloader-reset sonoff \
	--probe-method bootloader \
	flash --force --firmware your-firmware.gbl
```

aquova, 2024-08-15:
With that command, it doesn't get any farther unfortunately.

```
2024-08-15 07:00:33.448 sapphire asyncio DEBUG Using selector: EpollSelector
2024-08-15 07:00:33.448 sapphire universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.3', ezsp_version='7.4.3.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-08-15 07:00:33.448 sapphire universal_silabs_flasher.flash DEBUG Probing with 115200 baudrate first
2024-08-15 07:00:33.449 sapphire universal_silabs_flasher.flasher INFO Triggering sonoff bootloader
2024-08-15 07:00:33.449 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0' (baudrate=115200, xonxoff=False, rtscts=False)
2024-08-15 07:00:33.472 sapphire universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.common.SerialProtocol object at 0x7aaeb33388f0>, Serial<id=0x7aaeb33388b0, open=True>(port='/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-08-15 07:00:34.094 sapphire universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-08-15 07:00:34.094 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0' (baudrate=115200, xonxoff=False, rtscts=False)
2024-08-15 07:00:34.118 sapphire universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x7aaeb3338980>, Serial<id=0x7aaeb3338880, open=True>(port='/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e5449a9ca45ed118f6997a40a86e0b4-if00-port0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-08-15 07:00:34.118 sapphire universal_silabs_flasher.common DEBUG Sending data b'\n'
2024-08-15 07:00:34.118 sapphire universal_silabs_flasher.common DEBUG Sending data b'3'
Error: Failed to probe running application type
```

puddly, 2024-08-15:
Does your stick work at all? If not even the hardware bootloader triggers, it sounds like it is broken.

aquova, 2024-08-15:
It does work, but it's been on the fritz. I use it with Home Assistant + Zigbee2MQTT and over the past week or two, it's worked very sporadically, usually shortly after rebooting the server but not for very long. I was advised that upgrading the firmware might solve the issue, but I'm wondering if it's a hardware issue instead now.

aquova, 2024-08-19:
I picked up another one off of Amazon, tried the web interface then the command line one -- pretty much the exact same issue

```
2024-08-19 16:47:22.978 sapphire asyncio DEBUG Using selector: EpollSelector
2024-08-19 16:47:22.978 sapphire universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.3', ezsp_version='7.4.3.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-08-19 16:47:22.978 sapphire universal_silabs_flasher.flash DEBUG Probing app type ApplicationType.EZSP first
2024-08-19 16:47:22.978 sapphire universal_silabs_flasher.flash DEBUG Probing with 115200 baudrate first
2024-08-19 16:47:22.979 sapphire universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-08-19 16:47:22.979 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2285ae4462afed119526404e71c9e8b5-if00-port0' (baudrate=115200, xonxoff=False, rtscts=False)
2024-08-19 16:47:23.000 sapphire universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x7a73e8070530>, Serial<id=0x7a73e8070490, open=True>(port='/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2285ae4462afed119526404e71c9e8b5-if00-port0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-08-19 16:47:23.000 sapphire universal_silabs_flasher.common DEBUG Sending data b'\n'
2024-08-19 16:47:23.000 sapphire universal_silabs_flasher.common DEBUG Sending data b'3'
2024-08-19 16:47:25.010 sapphire universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-08-19 16:47:25.011 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2285ae4462afed119526404e71c9e8b5-if00-port0' (baudrate=115200, xonxoff=True, rtscts=False)
2024-08-19 16:47:25.041 sapphire bellows.ezsp DEBUG Resetting EZSP
2024-08-19 16:47:25.041 sapphire bellows.uart DEBUG Resetting ASH
2024-08-19 16:47:25.041 sapphire bellows.ash DEBUG Sending frame CANCEL + RstFrame() + FLAG
2024-08-19 16:47:25.041 sapphire bellows.ash DEBUG Sending data  1ac038bc7e
2024-08-19 16:47:30.044 sapphire universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2024-08-19 16:47:30.044 sapphire zigpy.serial DEBUG Opening a serial connection to '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2285ae4462afed119526404e71c9e8b5-if00-port0' (baudrate=460800, xonxoff=False, rtscts=False)
2024-08-19 16:47:30.061 sapphire bellows.uart DEBUG Connection lost: None
2024-08-19 16:47:30.061 sapphire bellows.uart DEBUG Closed serial connection
2024-08-19 16:47:30.061 sapphire universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0x7a73e8070ce0>, Serial<id=0x7a73e8070c70, open=True>(port='/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_2285ae4462afed119526404e71c9e8b5-if00-port0', baudrate=460800, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-08-19 16:47:30.061 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-19 16:47:30.062 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-19 16:47:31.062 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2024-08-19 16:47:31.163 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2024-08-19 16:47:31.163 sapphire universal_silabs_flasher.common DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2024-08-19 16:47:32.164 sapphire universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2024-08-19 16:47:32.264 sapphire universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
[... cut ...]
2024-08-19 16:47:46.381 sapphire universal_silabs_flasher.spinel DEBUG Sending frame SpinelFrame(header=SpinelHeader(transaction_id=3, network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02')
2024-08-19 16:47:46.381 sapphire universal_silabs_flasher.common DEBUG Sending data b'~\x83\x02\x02\xe65~'
2024-08-19 16:47:47.383 sapphire universal_silabs_flasher.spinel DEBUG Failed to send SpinelFrame(header=SpinelHeader(network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02'), trying again in 0.10s (attempt 4 of 4)
Error: Failed to probe running application type
```

aquova, 2024-08-19:
...ah. I didn't know that there were two different models. Both are the -P dongle

puddly, 2024-08-19:
That would explain things! The -P one uses a completely different chip and different firmware... Sorry, I should have spotted the model name in the serial port line from your log. They're nearly identical between the two very different models.

aquova, 2024-08-19:
That's alright, I didn't realize there was a distinction. Thanks for your time

closed 2024-08-19 by aquova