NabuCasa/universal-silabs-flasher #79: Unable to flash my Sonoff ZBdongle-E V1

ThibLeS-59, 2024-08-16:
I'm a beginner, and I can't seem to flash my ZBDongle-E V1. I first tried using the DARKXST site, where it was recognized, but the flashing failed. Then, I tried using the command line with universal_silabs_flasher, but the installation always ends in an error

Please help me 

Thanks

` File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Scripts\universal-silabs-flasher.exe\__main__.py", line 7, in <module>
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flash.py", line 40, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flash.py", line 341, in flash
    await flasher.probe_app_type()
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flasher.py", line 173, in probe_app_type
    await self.enter_bootloader_reset(self._reset_target)
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flasher.py", line 82, in enter_bootloader_reset
    await self.enter_serial_bootloader()
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flasher.py", line 86, in enter_serial_bootloader
    async with connect_protocol(self._device, baudrate, SerialProtocol) as sonoff:
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\common.py", line 181, in connect_protocol
    _, protocol = await zigpy.serial.create_serial_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\zigpy\serial.py", line 69, in create_serial_connection
    transport, protocol = await pyserial_asyncio.create_serial_connection(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial_asyncio\__init__.py", line 448, in create_serial_connection
    serial_instance = serial.serial_for_url(*args, **kwargs)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial\__init__.py", line 87, in serial_for_url
    instance = klass(None, *args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial\serialutil.py", line 233, in __init__
    self.exclusive = exclusive
    ^^^^^^^^^^^^^^
  File "C:\Users\TScourfield\AppData\Local\Programs\Python\Python312\Lib\site-packages\serial\serialwin32.py", line 475, in exclusive
    raise ValueError('win32 only supports exclusive access (not: {})'.format(exclusive))
ValueError: win32 only supports exclusive access (not: False)`

puddly, 2024-08-16:
Do you have a non-Windows computer you can use?

ThibLeS-59, 2024-08-16:
No, I only have Windows available, unfortunately

closed 2024-08-17 by puddly

ThibLeS-59, 2024-08-19:
Ok sorry. I'am beginner can you explain a manipulation ?

puddly, 2024-08-19:
Once we release a new version of `zigpy` in a day or so you will be able to upgrade it and the flasher should work.