NabuCasa/universal-silabs-flasher #91: TimeOut error when flashing via socket://ip:port

basilfx, 2024-11-14:
I have a ZigBee bridge that I connect to via TCP-over-serial.

When I try to flash a new firmware, I get this:

```
2024-11-14 23:37:59.451 MacBookPro.localdomain universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=2, sdk_version='4.4.4', ezsp_version='7.4.4.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant=None, baudrate=115200)
2024-11-14 23:37:59.451 MacBookPro.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-11-14 23:38:00.541 MacBookPro.localdomain universal_silabs_flasher.flasher INFO Detected bootloader version '2.4.2'
2024-11-14 23:38:00.541 MacBookPro.localdomain universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '2.4.2' at 115200 baudrate (bootloader baudrate 115200)
zb-gw03_zigbee_ncp_7.4.4.0.gbl  [------------------------------------]    0%
Traceback (most recent call last):
  File "/Users/basilfx/.local/pipx/venvs/universal-silabs-flasher/lib/python3.12/site-packages/universal_silabs_flasher/gecko_bootloader.py", line 71, in probe
    return await self.ebl_info()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/basilfx/.local/pipx/venvs/universal-silabs-flasher/lib/python3.12/site-packages/universal_silabs_flasher/gecko_bootloader.py", line 85, in ebl_info
    await self._state_machine.wait_for_state(State.IN_MENU)
  File "/Users/basilfx/.local/pipx/venvs/universal-silabs-flasher/lib/python3.12/site-packages/universal_silabs_flasher/common.py", line 119, in wait_for_state
    return await future
           ^^^^^^^^^^^^
```

It fails to probe prior flashing. But when I run the probe command, it works fine.

Here is another attempt, but with verbose logging:
```
2024-11-14 23:44:11.485 MacBookPro.localdomain asyncio DEBUG Using selector: KqueueSelector
2024-11-14 23:44:11.485 MacBookPro.localdomain universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=2, sdk_version='4.4.4', ezsp_version='7.4.4.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant=None, baudrate=115200)
2024-11-14 23:44:11.485 MacBookPro.localdomain universal_silabs_flasher.flash DEBUG Probing app type ApplicationType.EZSP first
2024-11-14 23:44:11.485 MacBookPro.localdomain universal_silabs_flasher.flash DEBUG Probing with 115200 baudrate first
2024-11-14 23:44:11.486 MacBookPro.localdomain universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-11-14 23:44:11.486 MacBookPro.localdomain zigpy.serial DEBUG Opening a serial connection to 'socket://10.0.0.13:8888' (baudrate=115200, xonxoff=False, rtscts=False)
2024-11-14 23:44:11.517 MacBookPro.localdomain zigpy.serial DEBUG Connection made: <_SelectorSocketTransport fd=3 read=idle write=<idle, bufsize=0>>
2024-11-14 23:44:11.517 MacBookPro.localdomain universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2024-11-14 23:44:11.518 MacBookPro.localdomain universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2024-11-14 23:44:11.518 MacBookPro.localdomain universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2024-11-14 23:44:11.575 MacBookPro.localdomain universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v2.04.02\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v2.04.02\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v2.04.02\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2024-11-14 23:44:11.575 MacBookPro.localdomain universal_silabs_flasher.gecko_bootloader DEBUG Detected version string '2.04.02'
2024-11-14 23:44:12.576 MacBookPro.localdomain zigpy.serial DEBUG Waiting for serial port to close
2024-11-14 23:44:12.576 MacBookPro.localdomain zigpy.serial DEBUG Connection lost: None
2024-11-14 23:44:12.576 MacBookPro.localdomain universal_silabs_flasher.flasher DEBUG Probe result: ProbeResult(version='2.4.2', continue_probing=None, baudrate=115200)
2024-11-14 23:44:12.577 MacBookPro.localdomain universal_silabs_flasher.flasher INFO Detected bootloader version '2.4.2'
2024-11-14 23:44:12.577 MacBookPro.localdomain universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '2.4.2' at 115200 baudrate (bootloader baudrate 115200)
zb-gw03_zigbee_ncp_7.4.4.0.gbl  [------------------------------------]    0%2024-11-14 23:44:12.579 MacBookPro.localdomain zigpy.serial DEBUG Opening a serial connection to 'socket://10.0.0.13:8888' (baudrate=115200, xonxoff=False, rtscts=False)
2024-11-14 23:44:12.586 MacBookPro.localdomain zigpy.serial DEBUG Connection made: <_SelectorSocketTransport fd=3 read=idle write=<idle, bufsize=0>>
2024-11-14 23:44:12.586 MacBookPro.localdomain universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2024-11-14 23:44:12.586 MacBookPro.localdomain universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2024-11-14 23:44:14.587 MacBookPro.localdomain zigpy.serial DEBUG Waiting for serial port to close
2024-11-14 23:44:14.587 MacBookPro.localdomain zigpy.serial DEBUG Connection lost: None
```

I modified [this](https://github.com/NabuCasa/universal-silabs-flasher/blob/dev/universal_silabs_flasher/gecko_bootloader.py#L78-L79) part as follows:

```python
self.send_data(b"\r\n")
self.send_data(GeckoBootloaderOption.EBL_INFO)
self.send_data(b"\r\n")
self.send_data(GeckoBootloaderOption.EBL_INFO)
self.send_data(b"\r\n")
```

Which seems to do the job, and now it successfully flashed the device (albeit it fails with 'universal_silabs_flasher.gecko_bootloader.NoFirmwareError: No firmware exists on the device', but another probe reveals the firmware is there).

I've been using this tool without problems in the past (I think v0.20.x), but that one could not handle the new image type (I built images based on https://github.com/basilfx/silabs-firmware-builder).

puddly, 2024-11-14:
It sounds like your device is stuck in the bootloader and has no valid application image. If that's the case, you need to run the flasher but override the probe methods so that it only tries to probe the bootloader and no other application types.

basilfx, 2024-11-15:
It wasn't stuck in the bootloader. I was using version 7.4.2.0 just fine. I tried flashing this two days ago, and did not succeed, so I restarted and everything was still fine. When manually connecting with the serial port, I could interact with it just fine.

I then tried to 'solve' it yesterday, and it eventually worked. Now I run 7.4.4.0.

closed 2025-01-11 by puddly