NabuCasa/universal-silabs-flasher #97: Flashing ZBDongle-E running router firmware fails

LinusCDE, 2025-02-03:
Trying to flash the SONOFF Dongle Plus E while using the router firmware fails. Afaik this should work.

Version: 0.0.28 (installed with pip today, 2025-02-03)
Command: `universal-silabs-flasher --bootloader-reset rts_dtr --device /dev/ttyUSB0 flash --firmware Downloads/ncp-uart-sw_EZNet7.4.4_V1.0.0.gbl`

<details>
<summary>Output (click to see)</summary>

```
2025-02-03 15:59:33.132 FrameLFK universal_silabs_flasher.flash INFO Failed to read firmware metadata: {exc!r}
2025-02-03 15:59:33.132 FrameLFK universal_silabs_flasher.flasher INFO Triggering rts_dtr bootloader
2025-02-03 15:59:33.788 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-02-03 15:59:33.926 FrameLFK universal_silabs_flasher.flasher INFO Launched application from bootloader
2025-02-03 15:59:33.938 FrameLFK universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2025-02-03 15:59:33.938 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-02-03 15:59:38.281 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2025-02-03 15:59:42.623 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2025-02-03 15:59:46.967 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2025-02-03 15:59:52.015 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.SPINEL at 460800 baud
2025-02-03 15:59:56.368 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.ROUTER at 115200 baud
2025-02-03 15:59:56.946 FrameLFK universal_silabs_flasher.flasher INFO Detected ApplicationType.ROUTER, version '6.10.3 GA build 297' (6.10.3.297) at 115200 baudrate (bootloader baudrate 115200)
Traceback (most recent call last):
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 288, in enter_bootloader
    await router.enter_bootloader()
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/router.py", line 107, in enter_bootloader
    await self._state_machine.wait_for_state(State.READY)
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/common.py", line 119, in wait_for_state
    return await future
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/linus/.local/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/usr/lib/python3.13/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.13/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3.13/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/flash.py", line 37, in inner
    return asyncio.run(f(*args, **kwargs))
           ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ~~~~~~~~~~^^^^^^
  File "/usr/lib64/python3.13/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/lib64/python3.13/asyncio/base_events.py", line 720, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/flash.py", line 422, in flash
    await flasher.enter_bootloader()
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 287, in enter_bootloader
    async with asyncio_timeout(PROBE_TIMEOUT):
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```
</details>

It seems to timeout out with the PROBE_TIMEOUT while trying to start the flash. I tried changing that timeout from 2 to 5 but it didn't fix it.

Also of interest is that provding `--probe-method router`, makes it fail pretty much instantly.
<details>

<summary>Output of that (click to expend)</summary>

```
2025-02-03 15:53:30.256 FrameLFK universal_silabs_flasher.flash WARNING The 'sonoff' reset target is deprecated. Use '--bootloader-reset rts_dtr' instead.
2025-02-03 15:53:30.257 FrameLFK universal_silabs_flasher.flash INFO Failed to read firmware metadata: {exc!r}
2025-02-03 15:53:30.257 FrameLFK universal_silabs_flasher.flasher INFO Triggering rts_dtr bootloader
2025-02-03 15:53:30.905 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.ROUTER at 115200 baud
Error: Failed to probe running application type
```

</details>

The command `probe` is able to detect it and read the version when trying ROUTER without issues.

---

No urgency from my side. I just though this might be some unknown regression that is useful to know. https://github.com/darkxst/silabs-firmware-builder/issues/172#issuecomment-2581367342 recommended this to flash the device when using the router firmware (as using [silabs-firmware-builder](https://github.com/darkxst/silabs-firmware-builder) fails to reset/detect the stick in router mode).

LinusCDE, 2025-02-03:
Sure, here you go:

Command: `universal-silabs-flasher -vvv --bootloader-reset rts_dtr --device /dev/ttyUSB0 flash --firmware Downloads/ncp-uart-sw_EZNet7.4.4_V1.0.0.gbl`

<details>

<summary>Output</summary>

```
2025-02-03 16:39:25.168 FrameLFK asyncio DEBUG Using selector: EpollSelector
2025-02-03 16:39:25.169 FrameLFK universal_silabs_flasher.flash INFO Failed to read firmware metadata: {exc!r}
2025-02-03 16:39:25.170 FrameLFK universal_silabs_flasher.flasher INFO Triggering rts_dtr bootloader
2025-02-03 16:39:25.170 FrameLFK zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB0' (baudrate=115200, xonxoff=False, rtscts=False)
2025-02-03 16:39:25.202 FrameLFK zigpy.serial DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.common.FlowControlSerialProtocol object at 0x7f51071dc590>, Serial<id=0x7f51071935e0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-03 16:39:25.203 FrameLFK universal_silabs_flasher.common DEBUG Setting UART signals: rts=1, cts=-, dtr=0
2025-02-03 16:39:25.306 FrameLFK universal_silabs_flasher.common DEBUG Setting UART signals: rts=0, cts=-, dtr=1
2025-02-03 16:39:25.811 FrameLFK universal_silabs_flasher.common DEBUG Setting UART signals: rts=0, cts=-, dtr=0
2025-02-03 16:39:25.815 FrameLFK zigpy.serial DEBUG Waiting for serial port to close
2025-02-03 16:39:25.827 FrameLFK zigpy.serial DEBUG Connection lost: None
2025-02-03 16:39:25.827 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-02-03 16:39:25.827 FrameLFK zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB0' (baudrate=115200, xonxoff=False, rtscts=False)
2025-02-03 16:39:25.851 FrameLFK zigpy.serial DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x7f51071dc590>, Serial<id=0x7f51071936a0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-03 16:39:25.851 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2025-02-03 16:39:25.852 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2025-02-03 16:39:25.858 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r')
2025-02-03 16:39:25.863 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. eb')
2025-02-03 16:39:25.863 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Detected version string '1.12.00'
2025-02-03 16:39:25.864 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.RUN_FIRMWARE
2025-02-03 16:39:25.867 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'l info\r\nBL > \x00')
2025-02-03 16:39:25.910 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'l info\r\nBL > \x00Reset info: 0x03 (EXT)\r\nExtended Reset info: 0x0301 (')
2025-02-03 16:39:25.962 FrameLFK universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'l info\r\nBL > \x00Reset info: 0x03 (EXT)\r\nExtended Reset info: 0x0301 (PIN)\r\n')
2025-02-03 16:39:25.964 FrameLFK universal_silabs_flasher.flasher INFO Launched application from bootloader
2025-02-03 16:39:25.964 FrameLFK zigpy.serial DEBUG Waiting for serial port to close
2025-02-03 16:39:25.975 FrameLFK zigpy.serial DEBUG Connection lost: None
2025-02-03 16:39:25.975 FrameLFK universal_silabs_flasher.flasher DEBUG Probe result: ProbeResult(version='1.12.0', continue_probing=True, baudrate=115200)
2025-02-03 16:39:25.976 FrameLFK universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2025-02-03 16:39:25.976 FrameLFK universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-02-03 16:39:25.976 FrameLFK zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB0' (baudrate=460800, xonxoff=False, rtscts=False)
2025-02-03 16:39:25.999 FrameLFK zigpy.serial DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0x7f51071dc1a0>, Serial<id=0x7f5107193be0, open=True>(port='/dev/ttyUSB0', baudrate=460800, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-03 16:39:25.999 FrameLFK universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-02-03 16:39:26.000 FrameLFK universal_silabs_flasher.cpc DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
[... cut ...]
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/flash.py", line 422, in flash
    await flasher.enter_bootloader()
  File "/home/linus/.local/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 287, in enter_bootloader
    async with asyncio_timeout(PROBE_TIMEOUT):
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

</details>

puddly, 2025-02-03:
Ah, I see the problem. You're using ITEAD's router firmware. It's ancient and doesn't support any way to re-enter the bootloader from software.

You don't need a software bootloader trigger because your stick supports a hardware bootloader trigger: `--bootloader-reset rts_dtr`.

puddly, 2025-02-03:
The issue arises because the flasher only ever considers using the hardware bootloader trigger if there is no software trigger. So far, the ITEAD router firmware is the only firmware that's unable to be software triggered (as in, you're stuck with the firmware that's running).

You can bypass this by instructing the flasher to not try probing the stick and just flash: `--probe-methods bootloader`.

Regarding more up-to-date firmwares: @darkxst was working on a router firmware (https://github.com/darkxst/silabs-firmware-builder/pull/152) but it doesn't look like one was ever built for Sonoff sticks.

LinusCDE, 2025-02-03:
Great! And thanks for the quick reponses!

I somehow overlooked that method in the help and assumed, I'd only need `router` anyway.

Using that works flawless and doesn't require opening the stick anymore. Huge!

`universal-silabs-flasher --probe-methods bootloader --bootloader-reset rts_dtr --device /dev/ttyUSB0 flash --firmware Downloads/ncp-uart-sw_EZNet7.4.4_V1.0.0.gbl`

[image]

I'm completely fine using that workaround. I'll drop the command in https://github.com/darkxst/silabs-firmware-builder/issues/172 as well, so others can find the solution easier. :+1:

closed 2025-02-03 by LinusCDE