NabuCasa/universal-silabs-flasher #99: SONOFF Zigbee Dongle-E, stuck in bootloader (?)

leezer3, 2025-02-13:
I'm somewhat stuck at the minute. I've probably overlooked something, but would like the input of fresh eyes....

I'm attempting to flash a Sonoff Zigbee Dongle-E to coordinator mode.

My current state appears to be stuck in what I think is the bootloader, with no application image booting. (solid red light on the dongle)
I've attempted flashing with both no probe options (complains it can't find the running application type) and with the bootloader probe option, which gives a timeout error.

Oddly, if I attempt to upload files via a raw serial connection, it completes, but I don't get a flash successful option, it just drops me back to the main bootloader menu. Attempting to boot just gives me garbage characters.

Anyone with more experience able to give me a pointer as to where I've gone wrong?

---

Probe output (appears to find the firmware info OK):
```
universal-silabs-flasher --device COM6 probe
2025-02-13 18:52:13.629 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-02-13 18:52:15.732 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-02-13 18:52:20.105 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2025-02-13 18:52:24.499 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2025-02-13 18:52:28.865 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2025-02-13 18:52:31.144 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '7.4.5.0 build 0' (7.4.5.0.0) at 115200 baudrate (bootloader baudrate None)
2025-02-13 18:52:31.144 DESKTOP-IDUIACU universal_silabs_flasher.flash INFO Dumping EmberZNet Config
CONFIG_PACKET_BUFFER_COUNT=255
CONFIG_NEIGHBOR_TABLE_SIZE=26
CONFIG_APS_UNICAST_MESSAGE_COUNT=32
CONFIG_BINDING_TABLE_SIZE=32
CONFIG_ADDRESS_TABLE_SIZE=16
CONFIG_MULTICAST_TABLE_SIZE=16
CONFIG_ROUTE_TABLE_SIZE=16
CONFIG_DISCOVERY_TABLE_SIZE=8
CONFIG_STACK_PROFILE=0
CONFIG_SECURITY_LEVEL=5
CONFIG_MAX_HOPS=30
CONFIG_MAX_END_DEVICE_CHILDREN=32
CONFIG_INDIRECT_TRANSMISSION_TIMEOUT=3000
CONFIG_END_DEVICE_POLL_TIMEOUT=8
CONFIG_TX_POWER_MODE=0
CONFIG_DISABLE_RELAY=0
CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=0
CONFIG_SOURCE_ROUTE_TABLE_SIZE=200
CONFIG_FRAGMENT_WINDOW_SIZE=1
CONFIG_FRAGMENT_DELAY_MS=0
CONFIG_KEY_TABLE_SIZE=12
CONFIG_APS_ACK_TIMEOUT=1600
CONFIG_ACTIVE_SCAN_DURATION=3
CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD=2
CONFIG_REQUEST_KEY_TIMEOUT=0
CONFIG_APPLICATION_ZDO_FLAGS=0
CONFIG_BROADCAST_TABLE_SIZE=15
CONFIG_MAC_FILTER_TABLE_SIZE=15
CONFIG_SUPPORTED_NETWORKS=1
CONFIG_SEND_MULTICASTS_TO_SLEEPY_ADDRESS=0
CONFIG_ZLL_GROUP_ADDRESSES=0
CONFIG_ZLL_RSSI_THRESHOLD=216
CONFIG_MTORR_FLOW_CONTROL=1
CONFIG_RETRY_QUEUE_SIZE=16
CONFIG_NEW_BROADCAST_ENTRY_THRESHOLD=9
CONFIG_TRANSIENT_KEY_TIMEOUT_S=300
CONFIG_BROADCAST_MIN_ACKS_NEEDED=255
CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S=300
CONFIG_CTUNE_VALUE=128
CONFIG_ASSUME_TC_CONCENTRATOR_TYPE=1
CONFIG_GP_PROXY_TABLE_SIZE=5
EZSP_CONFIG_GP_SINK_TABLE_SIZE=0
```

Flash with no probing options:
```
universal-silabs-flasher --device COM6 flash --firmware ncp-uart-hw-v7.4.5.0-zbdonglee-115200
[... 91 lines cut ...]
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flash.py", line 37, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 685, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flash.py", line 437, in flash
    await flasher.flash_firmware(
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flasher.py", line 324, in flash_firmware
    await gecko.probe()
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\gecko_bootloader.py", line 70, in probe
    async with asyncio_timeout(PROBE_TIMEOUT):
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\asyncio\timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

leezer3, 2025-02-13:
With bootloader probe:
```
universal-silabs-flasher -vvv --device COM6 --probe-method bootloader --bootloader-reset sono
ff flash --firmware ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl --force
2025-02-13 19:36:55.321 DESKTOP-IDUIACU universal_silabs_flasher.flash WARNING The 'sonoff' reset target is deprecated. Use '--bootloader-reset rts_dtr' instead.
2025-02-13 19:36:55.321 DESKTOP-IDUIACU asyncio DEBUG Using proactor: IocpProactor
2025-02-13 19:36:55.321 DESKTOP-IDUIACU universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.5', ezsp_version='7.4.5.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant=None, baudrate=115200)
2025-02-13 19:36:55.321 DESKTOP-IDUIACU universal_silabs_flasher.flash DEBUG Probing with 115200 baudrate first
2025-02-13 19:36:55.321 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Triggering rts_dtr bootloader
2025-02-13 19:36:55.325 DESKTOP-IDUIACU zigpy.serial DEBUG Opening a serial connection to 'COM6' (baudrate=115200, xonxoff=False, rtscts=False)
2025-02-13 19:36:55.423 DESKTOP-IDUIACU zigpy.serial DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.common.FlowControlSerialProtocol object at 0x0000000008F16330>, Serial<id=0x8f161d0, open=True>(port='COM6', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-13 19:36:55.423 DESKTOP-IDUIACU universal_silabs_flasher.common DEBUG Setting UART signals: rts=1, cts=-, dtr=0
2025-02-13 19:36:55.530 DESKTOP-IDUIACU universal_silabs_flasher.common DEBUG Setting UART signals: rts=0, cts=-, dtr=1
2025-02-13 19:36:56.039 DESKTOP-IDUIACU universal_silabs_flasher.common DEBUG Setting UART signals: rts=0, cts=-, dtr=0
2025-02-13 19:36:56.047 DESKTOP-IDUIACU zigpy.serial DEBUG Waiting for serial port to close
2025-02-13 19:36:56.062 DESKTOP-IDUIACU zigpy.serial DEBUG Connection lost: None
2025-02-13 19:36:56.062 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-02-13 19:36:56.062 DESKTOP-IDUIACU zigpy.serial DEBUG Opening a serial connection to 'COM6' (baudrate=115200, xonxoff=False, rtscts=False)
2025-02-13 19:36:56.164 DESKTOP-IDUIACU zigpy.serial DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x0000000008F15FA0>, Serial<id=0x8f16440, open=True>(port='COM6', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-13 19:36:56.164 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2025-02-13 19:36:56.164 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2025-02-13 19:36:56.174 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r')
2025-02-13 19:36:56.190 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2025-02-13 19:36:56.190 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Detected version string '1.12.00'
2025-02-13 19:36:56.190 DESKTOP-IDUIACU zigpy.serial DEBUG Waiting for serial port to close
2025-02-13 19:36:56.197 DESKTOP-IDUIACU zigpy.serial DEBUG Connection lost: None
2025-02-13 19:36:56.197 DESKTOP-IDUIACU universal_silabs_flasher.flasher DEBUG Probe result: ProbeResult(version='1.12.0', continue_probing=False, baudrate=115200)
2025-02-13 19:36:56.197 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2025-02-13 19:36:56.197 DESKTOP-IDUIACU universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.12.0' at 115200 baudrate (bootloader baudrate 115200)
ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl
2025-02-13 19:36:56.202 DESKTOP-IDUIACU zigpy.serial DEBUG Opening a serial connection to 'COM6' (baudrate=115200, xonxoff=False, rtscts=False)
2025-02-13 19:36:56.305 DESKTOP-IDUIACU zigpy.serial DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x0000000008F17EF0>, Serial<id=0x8f3e260, open=True>(port='COM6', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-13 19:36:56.305 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2025-02-13 19:36:56.305 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2025-02-13 19:36:57.308 DESKTOP-IDUIACU universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x1a\xc1\x02\x02\x9b{~')
2025-02-13 19:36:58.303 DESKTOP-IDUIACU zigpy.serial DEBUG Waiting for serial port to close
2025-02-13 19:36:58.319 DESKTOP-IDUIACU zigpy.serial DEBUG Connection lost: None
Traceback (most recent call last):
[... cut ...]
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\flasher.py", line 324, in flash_firmware
    await gecko.probe()
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\site-packages\universal_silabs_flasher\gecko_bootloader.py", line 70, in probe
    async with asyncio_timeout(PROBE_TIMEOUT):
  File "C:\Users\leezer3\AppData\Local\Programs\Python\Python312\Lib\asyncio\timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

leezer3, 2025-02-13:
If I'm reading the code correctly, both are dumping out at the same point, here:
https://github.com/NabuCasa/universal-silabs-flasher/blob/dev/universal_silabs_flasher/gecko_bootloader.py#L71

All that seems to be doing is attempting to read the actual version info from the console.

puddly, 2025-02-13:
It's erroring out because the bootloader isn't responsive the second time it connects. I haven't tested the flasher very much on Windows so it's possible there's a driver glitch when reconnecting, causing this problem.

Can you try this on macOS or Linux?

leezer3, 2025-02-13:
Thanks. 
I've got *partial* success with that. It allowed me to install the Router firmware from here:
https://www.zigbee2mqtt.io/devices/ZBDongle-E.html
(the v7 coordinator firmware installed but failed to boot)

The dongle now boots as a router on v6.10.3, which is progress.

However, after flashing that, it doesn't seem to want to get back into the bootloader if I'm reading it right.

Log from attempting to flash V7 coordinator is below.
It dumps out here: 
https://github.com/NabuCasa/universal-silabs-flasher/blob/730177040c22c64ed072f1cbceffd52f07884e05/universal_silabs_flasher/flasher.py#L288

If I'm reading that right, you're asking it to reboot to the bootloader, and it's not doing that.....
```
universal-silabs-flasher -vvv --device /dev/ttyUSB0 --bootloader-reset sonoff flash --firmware ncp-uart-sw_7.4.5.0_115200.gbl --force
2025-02-13 20:52:15.363 UNIMATRIX-001 universal_silabs_flasher.flash WARNING The 'sonoff' reset target is deprecated. Use '--bootloader-reset rts_dtr' instead.
2025-02-13 20:52:15.363 UNIMATRIX-001 asyncio DEBUG Using selector: EpollSelector
2025-02-13 20:52:15.364 UNIMATRIX-001 universal_silabs_flasher.flash INFO Failed to read firmware metadata: {exc!r}
2025-02-13 20:52:15.364 UNIMATRIX-001 universal_silabs_flasher.flasher INFO Triggering rts_dtr bootloader
2025-02-13 20:52:15.364 UNIMATRIX-001 zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB0' (baudrate=115200, xonxoff=False, rtscts=False)
2025-02-13 20:52:15.388 UNIMATRIX-001 zigpy.serial DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.common.FlowControlSerialProtocol object at 0x7efed4ea30e0>, Serial<id=0x7efed4e9e0b0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-13 20:52:15.389 UNIMATRIX-001 universal_silabs_flasher.common DEBUG Setting UART signals: rts=1, cts=-, dtr=0
2025-02-13 20:52:15.493 UNIMATRIX-001 universal_silabs_flasher.common DEBUG Setting UART signals: rts=0, cts=-, dtr=1
2025-02-13 20:52:15.998 UNIMATRIX-001 universal_silabs_flasher.common DEBUG Setting UART signals: rts=0, cts=-, dtr=0
2025-02-13 20:52:16.002 UNIMATRIX-001 zigpy.serial DEBUG Waiting for serial port to close
2025-02-13 20:52:16.017 UNIMATRIX-001 zigpy.serial DEBUG Connection lost: None
2025-02-13 20:52:16.017 UNIMATRIX-001 universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-02-13 20:52:16.018 UNIMATRIX-001 zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB0' (baudrate=115200, xonxoff=False, rtscts=False)
2025-02-13 20:52:16.043 UNIMATRIX-001 zigpy.serial DEBUG Connection made: SerialTransport(<_UnixSelectorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x7efed4ea2900>, Serial<id=0x7efed4e9e380, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-02-13 20:52:16.043 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2025-02-13 20:52:16.044 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2025-02-13 20:52:16.051 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r')
2025-02-13 20:52:16.057 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. eb')
2025-02-13 20:52:16.057 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Detected version string '1.12.00'
2025-02-13 20:52:16.057 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.RUN_FIRMWARE
2025-02-13 20:52:16.059 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'l info\r\nBL > \x00')
2025-02-13 20:52:16.105 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'l info\r\nBL > \x00Reset info: 0x03 (EXT)\r\nExtended Reset info: 0x0301 (')
2025-02-13 20:52:16.156 UNIMATRIX-001 universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'l info\r\nBL > \x00Reset info: 0x03 (EXT)\r\nExtended Reset info: 0x0301 (PIN)\r\n')
2025-02-13 20:52:16.159 UNIMATRIX-001 universal_silabs_flasher.flasher INFO Launched application from bootloader
2025-02-13 20:52:16.159 UNIMATRIX-001 zigpy.serial DEBUG Waiting for serial port to close
2025-02-13 20:52:16.173 UNIMATRIX-001 zigpy.serial DEBUG Connection lost: None
2025-02-13 20:52:16.173 UNIMATRIX-001 universal_silabs_flasher.flasher DEBUG Probe result: ProbeResult(version='1.12.0', continue_probing=True, baudrate=115200)
2025-02-13 20:52:16.174 UNIMATRIX-001 universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2025-02-13 20:52:16.174 UNIMATRIX-001 universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-02-13 20:52:16.174 UNIMATRIX-001 zigpy.serial DEBUG Opening a serial connection to '/dev/ttyUSB0' (baudrate=460800, xonxoff=False, rtscts=False)
[... cut ...]
  File "/home/christopher/.venv/lib/python3.13/site-packages/universal_silabs_flasher/flash.py", line 422, in flash
    await flasher.enter_bootloader()
  File "/home/christopher/.venv/lib/python3.13/site-packages/universal_silabs_flasher/flasher.py", line 287, in enter_bootloader
    async with asyncio_timeout(PROBE_TIMEOUT):
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

```

puddly, 2025-02-13:
The router firmware has no way to enter the bootloader, unfortunately. You would need to pass `--probe-method bootloader` to disable probing and to force it to just flash after entering the bootloader via RTS/DTR.

leezer3, 2025-02-14:
So, that's interesting.

I can get it into bootloader mode and flash another file with the little buttons.
This however fails to boot, and I'm back to square one.

Broken hardware?

puddly, 2025-02-14:
What GBL were you trying to flash?

If it boots into the bootloader, it can run code off of the flash. Either there is NVRAM corruption and a full erase is required or the wrong application image is being flashed.

Elecktro1337, 2025-03-23:
I am joining this issue. Absolutely similar symptoms. Regarding which GBL I stitched: yes, that's it. Currently, 7.4.5 is being determined.
Downgrade to 7.2.* or upgrade to 8.0.2 returns a Timeout error.
Manually loading bootloader through the buttons doesn't seem to work, in any case, I don't see any reaction.

puddly, 2025-03-23:
@Elecktro1337 please re-run the flasher with verbosity flags (`-vvv`) and post the log? Make sure nothing else is running that could be trying to communicate with the stick

Elecktro1337, 2025-03-23:
> [@Elecktro1337](https://github.com/Elecktro1337) please re-run the flasher with verbosity flags (`-vvv`) and post the log. Make sure nothing else is running that could be trying to communicate with the stick

`
PS C:\Users\eleck> universal-silabs-flasher -vvv --device COM6 --bootloader-reset sonoff flash --firmware C:\Users\eleck\Downloads\ncp-uart-sw_7.4.3.0_115200.gbl
2025-03-23 05:28:58.813 Pushman asyncio DEBUG Using proactor: IocpProactor
2025-03-23 05:28:58.815 Pushman universal_silabs_flasher.flash INFO Failed to read firmware metadata: {exc!r}
2025-03-23 05:28:58.815 Pushman universal_silabs_flasher.flasher INFO Triggering sonoff bootloader
2025-03-23 05:28:58.815 Pushman zigpy.serial DEBUG Opening a serial connection to 'COM6' (baudrate=115200, xonxoff=False, rtscts=False)
2025-03-23 05:28:58.911 Pushman zigpy.serial DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <zigpy.serial.SerialProtocol object at 0x000001A9982279B0>, Serial<id=0x1a998227850, open=True>(port='COM6', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-03-23 05:28:59.508 Pushman zigpy.serial DEBUG Waiting for serial port to close
2025-03-23 05:28:59.517 Pushman zigpy.serial DEBUG Connection lost: None
2025-03-23 05:28:59.517 Pushman universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-03-23 05:28:59.517 Pushman zigpy.serial DEBUG Opening a serial connection to 'COM6' (baudrate=115200, xonxoff=False, rtscts=False)
2025-03-23 05:28:59.608 Pushman zigpy.serial DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x000001A998227B90>, Serial<id=0x1a998227c10, open=True>(port='COM6', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-03-23 05:28:59.608 Pushman universal_silabs_flasher.gecko_bootloader DEBUG Sending data b'\n'
2025-03-23 05:28:59.608 Pushman universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2025-03-23 05:28:59.627 Pushman universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2025-03-23 05:28:59.627 Pushman universal_silabs_flasher.gecko_bootloader DEBUG Detected version string '1.12.00'
2025-03-23 05:28:59.627 Pushman universal_silabs_flasher.gecko_bootloader DEBUG Sending data GeckoBootloaderOption.RUN_FIRMWARE
2025-03-23 05:28:59.719 Pushman universal_silabs_flasher.flasher INFO Launched application from bootloader
2025-03-23 05:28:59.719 Pushman zigpy.serial DEBUG Waiting for serial port to close
2025-03-23 05:28:59.728 Pushman zigpy.serial DEBUG Connection lost: None
2025-03-23 05:28:59.728 Pushman universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2025-03-23 05:28:59.728 Pushman universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-03-23 05:28:59.728 Pushman zigpy.serial DEBUG Opening a serial connection to 'COM6' (baudrate=460800, xonxoff=False, rtscts=False)
2025-03-23 05:28:59.825 Pushman zigpy.serial DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.cpc.CPCProtocol object at 0x000001A998227C20>, Serial<id=0x1a998227f10, open=True>(port='COM6', baudrate=460800, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2025-03-23 05:28:59.825 Pushman universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-03-23 05:28:59.825 Pushman universal_silabs_flasher.cpc DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-03-23 05:29:00.830 Pushman universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2025-03-23 05:29:00.846 Pushman universal_silabs_flasher.cpc DEBUG Failed to parse buffer bytearray(b'\xf8'): ValueError('Invalid flag')
2025-03-23 05:29:00.923 Pushman universal_silabs_flasher.cpc DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-03-23 05:29:00.923 Pushman universal_silabs_flasher.cpc DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-03-23 05:29:01.919 Pushman universal_silabs_flasher.cpc DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, [... cut ...]
  File "C:\Users\eleck\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\universal_silabs_flasher\gecko_bootloader.py", line 71, in probe
    async with async_timeout.timeout(PROBE_TIMEOUT):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Lib\asyncio\timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
`

Elecktro1337, 2025-03-23:
What I want to mention:
I managed to connect to bootloader via ExtraPuTTY and download the firmware via xmodem.
If you pay attention, now Silicon defines 7.4.3 - the same firmware that I flashed.

However, when I click on 2 (Run) in ExtraPuTTY, I get some scribbles. The stick itself still has a red LED. Now I'll try to connect it to z2m. The initial symptom is that the permission join and further commands are not processed.

[image]

puddly, 2025-03-23:
> However, when I click on 2 (Run) in ExtraPuTTY, I get some scribbles

That's normal. Only the bootloader is ASCII, EmberZNet communicates with a binary protocol.

> UPD: Nahhh, also error in z2m.
z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Delivery failed for '41723'.'

That looks fine to me, the stick itself works, you just have device communication issues.

Elecktro1337, 2025-03-23:
> That looks fine to me, the stick itself works, you just have device communication issues.

Yes, you're right. I tried ZHA out of curiosity, the device connection works.
But the issue of changing the firmware remains relevant....

puddly, 2025-03-23:
Try running the flasher in a Linux environment, it might work better than Windows.